PX4/PX4-Autopilot #18568: Firmware 1.12.0 version to CUAV X7

jdom88, 2021-11-04:
**Describe the bug**
Using QGroundControl, when I load the firmware v1.12.0 in an CUAV X7, downloaded from https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.0, the version installed and detected is v1.11.2 not v1.12.0

**To Reproduce**
Steps to reproduce the behavior:
1. Download cuav_x7pro_default.px4 file from https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.0
2. Load this firmware in an CUAV X7 board using QGroundControl
3. Check firmware version installed using MAVLink Console
4. See versión 1.11.2, not 1.12.0

dagar, 2021-11-04:
It's probably just an issue with the tag.

Can you try the regular flashing through QGroundControl? By default it will give you the most recent stable version (currently v1.12.3).

jdom88, 2021-11-08:
Yes, with regular flashing through QGroundControl the version loaded is 1.12.3
Finally I obtained the version v1.12.0 compiling that version, "make cuav_x7pro_default", and then loaded by custom flashing through QGroundControl

dagar, 2021-11-09:
Is this resolved? I would stick with the latest stable v1.12.3 rather than an earlier tag.