PX4/PX4-Autopilot #18572: Commander doesn't handle system broadcast commands

lukegluke, 2021-11-05:
Shouldn't commander handle broadcast system commands here?
https://github.com/PX4/PX4-Autopilot/blob/f492fa9d3ab5a1e7b105594a1f637bd6a22bfad4/src/modules/commander/Commander.cpp#L743-L747

Like this:
```
	/* only handle commands that are meant to be handled by this system and component (or broadcast) */
	if (((cmd.target_system != _status.system_id) && (cmd.target_system != 0))
			|| ((cmd.target_component != _status.component_id)
					&& (cmd.target_component != 0))) {
		return false;
	}
```

As I see there are several module that publish internal vehicle_command (from_external = false), usually they use cleared  vehicle_command with zeroed target_system. How do they work then?

dagar, 2021-11-05:
@lukegluke yes I think you're right.

In general I think we need to more intentionally maintain a barrier between PX4 internal vehicle_commands and Mavlink commands.

FYI @julianoes

julianoes, 2021-11-05:
I agree, we should not ignore broadcast.

PR #18764 (2021-12-02): commander: handle system broadcast commands
This changes the command handling logic to accept commands which are targeted at any system, signaled by target_system set to 0.

As suggested by @lukegluke in #18572.

Closes #18572.
closed 2021-12-02 by dagar