PX4/PX4-Autopilot #18573: False positive "Start loiter with fixed bank angle"

Jaeyoung-Lim, 2021-11-05:
**Describe the bug**
Whenever doing a fixedwing takeoff, a warning message "Start loiter with fixed bank angle" is wrongly being triggered

**To Reproduce**
Steps to reproduce the behavior:
1. Start SITL with a plane model
`make px4_sitl gazebo_plane`
2. Start mission with a takeoff waypoint

**Log Files and Screenshots**
[image]

**Additional context**
- This seems to be appearing after https://github.com/PX4/PX4-Autopilot/pull/17547 has been merged

dagar, 2021-11-05:
What failsafe is actually triggering?

dagar, 2021-11-05:
Link loss confusion?

sfuhrer, 2021-11-05:
> What failsafe is actually triggering?

No failsafe actually triggers, it's just that `_pos_sp_triplet.current.valid ` in FW Position controller gets set to valid one iteration after `_control_mode.flag_control_auto_enabled` and `_control_mode.flag_control_position_enabled` are set to true, and caused the internal FW position mode to be set to FW_POSCTRL_MODE_AUTO_ALTITUDE for one brief moment after triggering the FW takeoff.
We could also think about removing `_pos_sp_triplet.current.valid `check from the logic here, I don't know in what case this can be invalidated.

dagar, 2021-11-05:
> We could also think about removing _pos_sp_triplet.current.valid check from the logic here, I don't know in what case this can be invalidated.

Originally I think it was more for the ability to explicitly publish something that invalidates existing position setpoints that consumers already have.

closed 2021-11-07 by Jaeyoung-Lim