PX4/PX4-Autopilot #18574: False positive Imbalanced propeller detected, landing advised for fixedwings

Jaeyoung-Lim, 2021-11-05:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Start SITL with a plane model
```
make px4_sitl gazebo_plane
```
2. Start mission with a takeoff waypoint

**Log Files and Screenshots**
```
INFO  [commander] Failsafe mode deactivated
INFO  [tone_alarm] notify positive
WARN  [fw_pos_control_l1] Start loiter with fixed bank angle.
INFO  [commander] Armed by external command
INFO  [fw_pos_control_l1] Takeoff on runway
INFO  [commander] Takeoff detected
INFO  [runway_takeoff] #Takeoff airspeed reached
WARN  [commander] Imbalanced propeller detected, landing advised
INFO  [runway_takeoff] #Climbout
```

**Additional context**
- The warning has been introduced from https://github.com/PX4/PX4-Autopilot/pull/18553

TSC21, 2021-11-05:
Seems that we need to reduce the threshold for thr fixed wing models in Gazebo. @bresch what do you say,

dagar, 2021-11-05:
What does vehicle_imu_status.var_accel look like in that log?

dagar, 2021-11-05:
By default `COM_IMB_PROP_ACT` is set to warning.

bresch, 2021-11-08:
@Jaeyoung-Lim Could you provide a log please?

Jaeyoung-Lim, 2021-11-08:
@bresch Here is one: https://logs.px4.io/plot_app?log=15f471bd-2a86-4855-9206-ce111cd0b393 :grinning:

bresch, 2021-11-08:
@Jaeyoung-Lim It's apparently because of the high horizontal acceleration when the pusher starts

[image]

We could
1) increase the threshold (quick fix)
2) make the detector less sensitive to quick changes (better)

dagar, 2021-11-09:
Fixed in https://github.com/PX4/PX4-Autopilot/pull/18590.

closed 2021-11-09 by dagar