PX4/PX4-Autopilot #18580: A Bug in MAV_CMD_LOGGING_START? (msgID 76 - command 2510)

BOB4Drone, 2021-11-06:
**Describe the bug**
There seems to be a `New-Delete-Type-Mismatch` bug in MAV_CMD_LOGGING_START. (msgID 76 - command 2510)
Whenever we send MAV_CMD_LOGGING_START packet after turning on the drone, there seems to be a problem in the `delete` process.
This bug does not necessarily lead to a crash, but the drone seems to be reacting abnormally.
We are not absolutely sure about this bug because it does not directly affect the actual drone operation. But the sanitizer detects it with an abnormal reaction so we just wanted to make sure and check this issue.

**To Reproduce**
1. Turn on the drone, without connecting to GCS. (GCS must be turned off to induce ack_timeout.)
2. send a MAV_CMD_LOGGING_START packet to the drone
3. New-Delete-Type-Mismatch occurs.

**Packet Used to Reproduce this packet**
```
fd21000000ffbe4c000000000000000000000000000000000000000000000000000000000000ce09010100a49a
```

**Log Files and Screenshots**
Log File (Simulator) : https://logs.px4.io/plot_app?log=6b880c53-9efb-443a-a06b-2d6ab581cd3e
Log File (px4_fmu-v5) : https://logs.px4.io/plot_app?log=8a1d12f3-06e0-48d2-9b02-515c8cca38f9
ScreenShots
[image]
- in simulator 

[image]
- in px4_fmu-v5

**Drone (please complete the following information):**
px4_fmu-v5

**Additional context**
The overall flow we analyzed for the MAV_CMD_LOGGING_START packet is as follows.

***Process of making instance***
1. [MavlinkReceiver::handle_message_command_both](https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L696-L714)
2. [try_start_ulog_streaming](https://github.com/PX4/PX4-Autopilot/blob/6d78054f5001548ea2f1d48bb59966e2352bc9fe/src/modules/mavlink/mavlink_main.h#L506-L511)
3. [MavlinkULog::try_start](https://github.com/PX4/PX4-Autopilot/blob/54c91002cdb7c6d97f6601e087a247a415854056/src/modules/mavlink/mavlink_ulog.cpp#L210-L232)

***Process of closing the logger***
1. [Mavlink::task_main](https://github.com/PX4/PX4-Autopilot/blob/6d78054f5001548ea2f1d48bb59966e2352bc9fe/src/modules/mavlink/mavlink_main.cpp#L2468-L2490)
2. [MavlinkULog::handle_update](https://github.com/PX4/PX4-Autopilot/blob/54c91002cdb7c6d97f6601e087a247a415854056/src/modules/mavlink/mavlink_ulog.cpp#L109-L112)
   - this function returns -110(-ETIMEDOUT) because GCS is not connected.
3. [MavlinkULog::stop](https://github.com/PX4/PX4-Autopilot/blob/54c91002cdb7c6d97f6601e087a247a415854056/src/modules/mavlink/mavlink_ulog.cpp#L234-L245)

https://github.com/PX4/PX4-Autopilot/blob/54c91002cdb7c6d97f6601e087a247a415854056/src/modules/mavlink/mavlink_ulog.cpp#L234-L245
There seems to be a problem when deleting the instance. 
We are assuming that a logger instance should not be made by the packet in the first place because a logger file is already made initially.

This is what we have got so far, and we would love to hear back from you.

bkueng, 2021-11-08:
Hmm ASan doesn't event want to start for me (`AddressSanitizer: SEGV on unknown address 0xffffffffffffffff`).
And valgrind tells me everything is fine.

Is `MAV_CMD_LOGGING_START` the only command you send? And only once? Could you test with valgrind as well?

> This bug does not necessarily lead to a crash, but the drone seems to be reacting abnormally.

Can you elaborate on this?

BOB4Drone, 2021-11-08:
1. The build command I used for this issue is `PX4_ASAN=1 make px4_sitl gazebo PX4_BUILD_TYPE=0`. Maybe re-cloning from the git might solve the issue? Oh and ASAN only works on simulator by the way. (ASAN doesn't work on the actual px4 board)

2. Yes, I've sent MAV_CMD_LOGGING_START only once. The following is what i've done (based on simulator build)
- start up the drone with `PX4_ASAN=1 make px4_sitl gazebo PX4_BUILD_TYPE=0`
- send the packet to the corresponding IP with port 18570 (it is the port used for simulators)
- bug detected in about several seconds.
I've also uploaded the crash process on youtube, so check if you want to : https://www.youtube.com/watch?v=xkS3w3dbUhc
And hmm... I've checked on valgrind too, and yes, it does not show anything to me either..

3. More explanation on "This bug does not necessarily lead to a crash, but the drone seems to be reacting abnormally." part
You can just ignore this part. I just wanted to say that sending a single packet does not shut down the simulator.(without ASAN)

bkueng, 2021-11-09:
I don't want to dismiss this lightly, and you're welcome to dig further in the code, if you find something, but given the information, I'd have to say this is a false positive by the sanitizer.

BOB4Drone, 2021-11-09:
Okay I see what you are saying. Thanks for the response, and we'll tell you if we find something else.
Have a good day!

closed 2021-11-24 by BOB4Drone