PX4/PX4-Autopilot #18585: jMavSim Running Error

DuyTLe7, 2021-11-08:
Hi, 
I am following the installation of PX4 toolchain, and I ran into an error when trying to run jMavSim using the command: make px4_sitl jmavsim

Here is the error:
$ make px4_sitl jmavsim
-- PX4 version: v1.13.0-alpha1-2883-g66ab7e50df
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3")
-- PX4 config file: /cygdrive/c/PX4/home/PX4-Autopilot/PX4-Autopilot/boards/px4/sitl/default.px4board
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'menuconfig'
CMake Error at cmake/kconfig.cmake:6 (message):
  kconfiglib is not installed or not in PATH

  please install using "pip3 install kconfiglib"

Call Stack (most recent call first):
  CMakeLists.txt:168 (include)

-- Configuring incomplete, errors occurred!
Error: /cygdrive/c/PX4/home/PX4-Autopilot/PX4-Autopilot/build/px4_sitl_default is not a directory
make: *** [Makefile:230: px4_sitl] Error 1

Please let me know if you ran into the same error before and how to fix this!
Thank you!

liluyi, 2021-11-10:
You can try:
python3 -m pip install --user kconfiglib
instead of
pip3 install kconfiglib.

This may be caused by multiple python3 versions installed in your OS.

Beau3n, 2025-01-05:
> You can try: python3 -m pip install --user kconfiglib instead of pip3 install kconfiglib.
> 
> This may be caused by multiple python3 versions installed in your OS.

this really works ,thanks

closed 2025-10-29 by hamishwillee