PX4/PX4-Autopilot #18595: Fixed-wing landing consistently veering off to right

thomasteisberg, 2021-11-09:
**Describe the bug**
(Presumably this is user error and not a real bug, but I was asked to create an issue so I'll try to follow the template.)

The aircraft tracks the planned path quite well until approximately when it exits the landing loiter for the final approach. At this point, it seems to make a turn to the right and then land on this new path.

I would like to understand why this occurs and ideally how I can correct it.

**To Reproduce**
See logs below. We're flying entirely in mission mode (with the exception of noted aborted landings).

**Expected behavior**
I expected the plane to track the planned landing direction, heading on a path towards the landing waypoint.

**Log Files and Screenshots**

Original example: https://logs.px4.io/plot_app?log=eea3a1e0-be68-470d-98ab-3730bf357679
[image]
The divergence can be seen quite clearly in the above PX4 screenshot.
Flight plan: [nov3-flight3.plan.txt](https://github.com/PX4/PX4-Autopilot/files/7500669/nov3-flight3.plan.txt)

Looking at the yaw angle plot for this one, it looks to me like the plane is lagging behind the set point consistently until suddenly the set point is updated to approximately the current estimated yaw. 
[image]

Another example from this morning: https://logs.px4.io/plot_app?log=63428e7b-2bd6-490b-9d14-e347effb02a4
This one I had `FW_LND_HHDIST` set to 0.
There are two landing attempts on this one. The first showed the same behavior of veering off to the right and was going to miss our landing area entirely, so I switched to position control mode and we put it up in a hold. We uploaded a new plan with the landing pattern shifted significantly to the west and let it land on that one. Probably easier to decipher what's happening on the first log.

**Drone (please complete the following information):**
X-UAV Talon (the 1.7 m wingspan one -- not the mini or pro)
Autopilot hardware is a Pixhawk 4 mini with the GPS puck and a Lidar Lite v3.
Mounting of GPS puck and autopilot are as shown below:
[image]
[image]
(You can ignore the extra wires in the second picture - they were for some ground testing.)

Both are oriented in their normal direction.

I haven't done any tuning for this platform. I'm just flying it on the X-UAV Mini Talon profile, which seems to be plenty good enough to get it to fly, however I think it's quite clear from the logs that some of the control loops could be better tuned.

dagar, 2021-11-09:
Thanks for the detailed report with logs, I'll take a closer look.

sfuhrer, 2021-11-11:
It's reproducible in SITL.

Something is weird with the calculation of the bearing_lastwp_currwp. The prev_wp that's used for that is located on the loiter on the left side, instead of being at the exit of the loiter.  That's why the bearing_lastwp_currwp  is then not at 0, but at a few degrees.
[image]
[image]

thomasteisberg, 2021-11-11:
Thanks for reproducing it @sfuhrer! One thing I know we changed was using a counter-clockwise loiter. I tried recreating this in simulation and it seems like it might be the explanation for the waypoint issue you've identified.

This is what happens with the `gazebo_plane` simulation with the (default) clockwise landing loiter:

[image]
Flight plan: [test_cw.plan.txt](https://github.com/PX4/PX4-Autopilot/files/7522975/test_cw.plan.txt)

This is what happens when I change nothing except for unchecking the "Loiter clockwise" setting in the landing pattern:

[image]
Flight plan: [test_ccw.plan.txt](https://github.com/PX4/PX4-Autopilot/files/7522976/test_ccw.plan.txt)

Stab in the dark here, but it looks like something in PX4 is unaware of the clockwise/counter-clockwise setting so `bearing_lastwp_currwp` is always based on the exit location for a clockwise loiter?

Our next opportunity to fly is Monday. We'll see what happens with a clockwise loiter. Any other thoughts on what we should (or shouldn't) try with a real vehicle would also be appreciated!

Thanks for your help!

thomasteisberg, 2021-11-15:
We flew again today (with a real vehicle). We used a clockwise landing pattern and everything behaved as expected. I suppose this doesn't add much since we've already established the issue in SITL, but at least we have a workaround for now (only use clockwise loiter holds).

sfuhrer, 2021-11-24:
I've spent some time today trying to find the root cause, but beside confirming that it happens for every landing with a CCW loiter I've not found anything worth to mention.

Somehow the previous wp is being set to something that was never a "current" setpoint. It's always on the loiter circle though.
[image]

I don't know when I have time again for it.

ryanjAA, 2021-12-02:
@sfuhrer this has to do with proximity (l1 and acceptable radius possibly). If you take this exact plan and move the landing point by 30-40m rather say then with the l1 radius essentially still in the loiter to alt point, it lands where it should. That said, this is a strange edge case but it does work

ryanjAA, 2022-04-09:
Was this ever fixed (or confirmed fixed)?

LorenzMeier, 2022-04-09:
There is an ongoing rework of runway support by Thomas - so independent of this there is a whole set of changes coming.

ryanjAA, 2022-04-10:
Cool. Will be great to checkout.

sfuhrer, 2022-04-27:
Should be fixed by https://github.com/PX4/PX4-Autopilot/pull/19487

closed 2022-04-27 by sfuhrer