PX4/PX4-Autopilot #18596: H7 bootloader usart uploader : no respond

chuangniu1, 2021-11-09:
hardware:  cubeorange
software:    release/1.12
bootloader/defconfig: usart3 config

hw_config.h ->#define INTERFACE_USART_CONFIG         "/dev/ttyS0,115200"

if I want to upload .px4 file by usart2 ,I need change 
bootloader/defconfig: usart3 ->usart2

if I want to upload .px4 file by usart3,i need change
bootloader/defconfig: add usart2 usart3
hw_config.h-> /dev/ttyS0. change to /dev/ttyS1

If I use the release/1.11 branch and use stlink to burn the Release /1.11 bootloader, I find that the program fails to run and burn the Release /1.12 branch bootloader, Then burn release/1.11 cubepilot_cubeorange._default. px4 to run properly.
If I want to change the serial port of the Bootloader to serial port 3, I find it is still in the bootloader state even after the above changes

chuangniu1, 2021-11-09:
H7 chip bootloader compilation process, and how to sort out the serial port configuration? Just looking at bootloader/defconfig does not seem to satisfy the need to change the serial port
[https://github.com/PX4/PX4-Autopilot/blob/master/boards/cubepilot/cubeorange/src/hw_config.h](url)
[https://github.com/PX4/PX4-Autopilot/blob/master/boards/cubepilot/cubeorange/nuttx-config/bootloader/defconfig](url)

dagar, 2021-11-09:
The bootloader is built with `make cubepilot_cubeorange_bootloader`. 

Can you try running NuttX menuconfig (or qconfig) to add the UART?

``` Console
make cubepilot_cubeorange_bootloader menuconfig
```

chuangniu1, 2021-11-09:
[image]

I make Nuttx menuconfig, to add usart 1 usart 2.
issues is above

https://github.com/PX4/PX4-Autopilot/issues/18596#issue-1048147957

chuangniu1, 2021-12-20:
release/1.12 The serial port upgrade is normal. The reason for the previous failure is that the serial port is connected to the RS232 module, and the power supply circuit of the module is faulty
So this issue will be closed, thanks for your reply

closed 2021-12-20 by chuangniu1