PX4/PX4-Autopilot #18597: Cube Orange: console command "gps start" without additional parameters hangs autopilot

MikeCharikov, 2021-11-09:
Versions 1.12.3 (1.11 as well)

Console command "gps start" without additional parameters hangs Cube Orange. Same command on Cube Black starts GPS without problems. 

It is not a critical issue, but it may be a result of more serious issue.

[log_172_2021-11-9-09-14-30.zip](https://github.com/PX4/PX4-Autopilot/files/7503065/log_172_2021-11-9-09-14-30.zip)

dagar, 2021-11-11:
It's likely the default GPS device here. https://github.com/PX4/PX4-Autopilot/blob/e9d50b853ade5b3598ea461323978bcc8a60c96b/src/drivers/gps/gps.cpp#L1257

We should update it so that you're required to enter the device rather than a somewhat arbitrary default that's invalid in many cases.

dagar, 2021-11-11:
@MikeCharikov could you try https://github.com/PX4/PX4-Autopilot/pull/18620?

closed 2021-12-06 by bkueng