PX4/PX4-Autopilot #18606: Include board types in emitted MAVLink AUTOPILOT_VERSION.board_version

hamishwillee, 2021-11-09:
This follows on from [this change to the message definition](https://github.com/mavlink/mavlink/pull/1727). With this information a GCS can validate parameters for a particular board, potentially modify UI elements based on the shape or capabilities of the board, better determine what firmware updates are relevant.

@dagar you committed to do this in https://github.com/mavlink/mavlink/pull/1727#issuecomment-953931250

@julianoes If that goes in, perhaps something for MAVSDK to be aware of too?

julianoes, 2021-11-11:
@hamishwillee thanks for the note. I guess this is just exposed raw in MAVSDK:
https://github.com/mavlink/MAVSDK/blob/1fd81d3db220b8b65133534ade219d1c16e93a37/src/core/system.h#L205-L210

Note sure if I should change it or leave as is.

hamishwillee, 2025-10-28:
Looking at the code that sets [`AUTOPILOT_VERSION.board_version`](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION)

1. PX4 populates this in https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/mavlink/mavlink_main.cpp#L1091C23-L1091C40: 
   ```cpp
   msg.board_version = px4_board_version();
   ```
2. The implementation of `px4_board_version()` in https://github.com/PX4/PX4-Autopilot/blob/main/src/lib/version/version.c#L292-L299 is
   ```cpp
    uint32_t px4_board_version(void)
	{
	#if defined(__PX4_NUTTX)
		return CONFIG_CDCACM_PRODUCTID;
	#else
		return 1;
	#endif
	}
    ```
3. The `CONFIG_CDCACM_PRODUCTID` can come from either 
   - board definition such as `#define CONFIG_CDCACM_PRODUCTID 60` in https://github.com/PX4/PX4-Autopilot/blob/main/boards/auterion/fmu-v6s/src/board_config.h#L176 - which seems to map correctly to https://github.com/PX4/PX4-Bootloader/blob/main/board_types.txt#L22
   - A defconfig for a board such as https://github.com/PX4/PX4-Autopilot/blob/main/boards/px4/fmu-v4/nuttx-config/nsh/defconfig#L74C1-L74C31 [more commonly]

4. In MAVLink [`AUTOPILOT_VERSION.board_version`](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION) states: 
HW / board version (last 8 bits should be silicon ID, if any). The first 16 bits of this field specify a board type from an enumeration stored at https://github.com/PX4/PX4-Bootloader/blob/master/board_types.txt and with extensive additions at https://github.com/ArduPilot/ardupilot/blob/master/Tools/AP_Bootloader/board_types.txt

PX4 just copies the board type into the field, so this will be in the bottom whatever bits, and omit the silicon ID. A MAVLink compliant system will be getting 0 for the board type and falsely reading the silicon type.

---

What I think we need to do is modify PX4 populates this in https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/mavlink/mavlink_main.cpp#L1091C23-L1091C40:  to bit shift the value before we copy it in. But I am wondering how much that will break.

@DonLakeFlyer @julianoes Thoughts?
- If I make the bit shift who do we need to tell?
- What is the silicon ID?