PX4/PX4-Autopilot #18622: Don't support ubuntu 16.04 ?

nm46nm, 2021-11-11:
I just use ubuntu 16.04 and have cloned Firmware. 
But when I 
`source Tools/setup/ubuntu.sh`
it warning that ' ubuntu 16.04 is no longer supported'.
Really? Don't support 16.04 ?
And How can I make it work in 16.04 ?  Any old version of PX4 or ... ?
I don't want to upgrade ubuntu since I have something that must need 16.04.

Jaeyoung-Lim, 2021-11-12:
@nm46nm I believe Ubuntu 16.04 has become EOL(End of Life) since April 2021 (https://ubuntu.com/16-04)

nm46nm, 2021-11-13:
@Jaeyoung-Lim Thank you, but how can I clone an older version of PX4? and which version will work?

Jaeyoung-Lim, 2021-11-13:
@nm46nm You can checkout the release page of this repository, or if you want to checkout a specific commit you can have a look how you can do this here: https://www.atlassian.com/git/tutorials/using-branches/git-checkout

closed 2021-11-13 by Jaeyoung-Lim