PX4/PX4-Autopilot #18629: Offboard mode with EKF2_AID_MASK = 24

gitsr-sys, 2021-11-11:
**Describe the bug**
`EKF2_AID_MASK` is set as 24 (VIO only). And `COM_RC_OVERRIDE` is set as 1.
I saw that moving RC stick switches to Position mode with `COM_RC_OVERRIDE=1`
However, I was not able to override offboard mode by moving RC stick.
I am wondering if I can use Position mode with `EKF2_AID_MASK=24`.

Another information is that my controller setting has two switches for the flight mode and offboard mode.
At that time, the offboard mode switch was on, and I moved flight mode switch from stabilize to altitude and position modes repeatedly.

**Expected behavior**
When I move RC stick during offboard mode with `EKF2_AID_MASK=24 (VIO only)`, it supposed to switch to Position mode with `COM_RC_OVERRIDE=1`.

Is there anyone who can explain this issue?

Jaeyoung-Lim, 2021-11-13:
@gitsr-sys You need to send setpoints in order to enter offboard mode. Otherwise there is no setpoint to track and it doesn't make sense to go into offboard mode

gitsr-sys, 2021-11-15:
@Jaeyoung-Lim 
I sent setpoints, and it went to offboard mode.

My main question is it is possible to use Position mode with `EKF2_AID_MASK=24`.

Jaeyoung-Lim, 2021-11-15:
@gitsr-sys The EKF modes are orthogonal to offboard mode. Therefore yes you can

antonio-sc66, 2021-11-16:
@gitsr-sys  Hi, I've tested `COM_RC_OVERRIDE ` set to 1 and if you're into let's say Hold mode and move the sticks it switches to position but when the drone is in Offboard and the stick is moved it doesn't override it and enter Position, it always stays in Offboard.

@Jaeyoung-Lim The documentation states that `COM_RC_OVERRIDE ` should work in auto and/or offboard, I think there's indeed a bug with that feature

MaEtUgR, 2021-11-17:
> And COM_RC_OVERRIDE is set as 1.
> I was not able to override offboard mode by moving RC stick.

Please read the [documentation of the parameter `COM_RC_OVERRIDE`](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#COM_RC_OVERRIDE). It is a bitmask and setting the value to 1 will only set the first bit and hence enable the feature for autonomous modes like Takeff, Mission, Land, Return excluding Offboard. To enable it in Offboard mode you have to set the second bit so the value needs to be 3 (11 in binary). 

https://docs.px4.io/master/en/advanced_config/parameter_reference.html#COM_RC_OVERRIDE

It also shows the individual options conveniently if you use the latest QGC build available here: https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html

[image]

Since you're not the first person running into this "trap" I suggest we just enable RC override also for offboard by default. It can then be disabled if desired.

closed 2024-05-24 by gitsr-sys