PX4/PX4-Autopilot #18641: Firmware Build error: Nuttx cmath: FMU target compilation error

thomaspj1017, 2021-11-14:
**Describe the bug**
When trying to build the firmware for px4_fmu-v5, I'm running into a -Wfatal-error because of the use of __ULong in Nuttx cmath. 
PX4 Version/setup:
-- PX4 version: v1.13.0-alpha1-2821-g8f6fd5f37b
-- PX4 config file: /home/pthomas/src/Firmware/PX4-Autopilot/boards/px4/io-v2/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m3
-- CONSTRAINED_FLASH y
-- PX4 config: px4_io-v2_default
-- PX4 platform: nuttx

**To Reproduce**
I ran the bash ubuntu.sh for setup, rebooted the computer, and then tried 'make px4_fmu-v5_default' and ran into the cmath error.

**Expected behavior**
I expected the firmware to build

Error from terminal:

In file included from ../../../../../src/lib/mathlib/math/Limits.hpp:43:0,
                 from ../../../../../src/lib/mathlib/mathlib.h:44,
                 from ../../../../../src/lib/rc/sbus.cpp:53:
/usr/include/newlib/math.h:18:3: error: '__ULong' does not name a type
   __ULong i[2];
   ^~~~~~~

thomaspj1017, 2021-11-15:
Forgot to mention running on Ubuntu 18.04.

mrpollo, 2021-11-17:
@thomaspj1017 did you also pull the submodules? what compiler version are you using?

vegaf1, 2022-08-11:
same error

mrpollo, 2022-08-11:
Can you share more details @vegaf1 ?

vegaf1, 2022-08-12:
Ended up fixing the error by sourcing the ROS 2 environment @mrpollo. However I still have issues with building gazebo. The issue is listed below: 

https://github.com/PX4/PX4-Autopilot/issues/20015

closed 2025-11-27 by github-actions[bot]