PX4/PX4-Autopilot #18643: make emlid_navio2 issues in raspberry pi 4 model B

sandeshneupane, 2021-11-15:
**Describe the bug**
I was trying to build px4 for navio2 in raspberry pi 4 model b,
I am getting errors as follows after running -> make emlid_navio2 

I followed some previously reported issues link https://github.com/PX4/PX4-Autopilot/issues/11552

Still getting errors after following all suggested steps. 

Your help will be highly appreciated.
Thanks

error--

[ 96%] Building CXX object src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/MPU9250_AK8963.cpp.o
In file included from /home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic_bitset.h:36,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceMaster.hpp:5,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceNode.hpp:37,
from /home/pi/PX4-Autopilot/platforms/common/uORB/PublicationMulti.hpp:4,
from /home/pi/PX4-Autopilot/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:39,
from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/icm20948/ICM20948.hpp:46,
from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/icm20948/ICM20948.cpp:34:
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘class px4::atomic’:
/home/pi/PX4-Autopilot/src/drivers/imu/invensense/icm20948/ICM20948.hpp:172:51: required from here
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h:76:41: **error: static assertion failed: atomic is not lock-free for the given type T
static_assert(__atomic_always_lock_free(sizeof(T), 0), "atomic is not lock-free for the given type T");
~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.**
make[3]: *** [src/drivers/imu/invensense/icm20948/CMakeFiles/drivers__imu__invensense__icm20948.dir/build.make:63: src/drivers/imu/invensense/icm20948/CMakeFiles/drivers__imu__invensense__icm20948.dir/ICM20948.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:6616: src/drivers/imu/invensense/icm20948/CMakeFiles/drivers__imu__invensense__icm20948.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
In file included from /home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic_bitset.h:36,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceMaster.hpp:5,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceNode.hpp:37,
from /home/pi/PX4-Autopilot/platforms/common/uORB/PublicationMulti.hpp:4,
from /home/pi/PX4-Autopilot/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:39,
from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250_I2C.hpp:46,
from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250_I2C.cpp:34:
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘class px4::atomic’:
/home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250_I2C.hpp:139:51: required from here
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h:76:41: error: static assertion failed: atomic is not lock-free for the given type T
static_assert(__atomic_always_lock_free(sizeof(T), 0), "atomic is not lock-free for the given type T");
~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250_i2c.dir/build.make:63: src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250_i2c.dir/MPU9250_I2C.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from /home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic_bitset.h:36,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceMaster.hpp:5,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceNode.hpp:37,
from /home/pi/PX4-Autopilot/platforms/common/uORB/PublicationMulti.hpp:4,
from /home/pi/PX4-Autopilot/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:39,
from /home/pi/PX4-Autopilot/src/drivers/imu/analog_devices/adis16448/ADIS16448.hpp:47,
from /home/pi/PX4-Autopilot/src/drivers/imu/analog_devices/adis16448/ADIS16448.cpp:34:
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘class px4::atomic’:
/home/pi/PX4-Autopilot/src/drivers/imu/analog_devices/adis16448/ADIS16448.hpp:115:51: required from here
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h:76:41: error: static assertion failed: atomic is not lock-free for the given type T
static_assert(__atomic_always_lock_free(sizeof(T), 0), "atomic is not lock-free for the given type T");
~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/build.make:63: src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/ADIS16448.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
[ 96%] Building CXX object src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250_i2c.dir/mpu9250_i2c_main.cpp.o
[ 96%] Building CXX object src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/adis16448_main.cpp.o
[ 96%] Building CXX object src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/MPU9250.cpp.o
In file included from /home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic_bitset.h:36,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceMaster.hpp:5,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceNode.hpp:37,
from /home/pi/PX4-Autopilot/platforms/common/uORB/PublicationMulti.hpp:4,
from /home/pi/PX4-Autopilot/src/lib/drivers/magnetometer/PX4Magnetometer.hpp:38,
from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250_AK8963.hpp:47,
from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250_AK8963.cpp:34:
[... 30 lines cut ...]
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘class px4::atomic’:
/home/pi/PX4-Autopilot/src/drivers/imu/analog_devices/adis16448/ADIS16448.hpp:115:51: required from here
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h:76:41: error: static assertion failed: atomic is not lock-free for the given type T
static_assert(__atomic_always_lock_free(sizeof(T), 0), "atomic is not lock-free for the given type T");
~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/build.make:76: src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/adis16448_main.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:6511: src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/all] Error 2
In file included from /home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic_bitset.h:36,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceMaster.hpp:5,
from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceNode.hpp:37,
from /home/pi/PX4-Autopilot/platforms/common/uORB/PublicationMulti.hpp:4,
from /home/pi/PX4-Autopilot/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:39,
from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250.hpp:46,
from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250.cpp:34:
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘class px4::atomic’:
/home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250.hpp:153:51: required from here
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h:76:41: error: static assertion failed: atomic is not lock-free for the given type T
static_assert(__atomic_always_lock_free(sizeof(T), 0), "atomic is not lock-free for the given type T");
~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/build.make:76: src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/MPU9250.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:6724: src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/all] Error 2
make[1]: *** [Makefile:152: all] Error 2
make: *** [Makefile:230: emlid_navio2] Error 2

mrpollo, 2021-11-17:
Can you describe what the environment you are trying to build this on looks like? What OS (name/version) and which compiler (name/version) are you trying to build on.

As a helpful tip, you can build any target from a different machine from what you intend to run this on.

sandeshneupane, 2021-11-19:
I am using Debian 10.4. 
:~ $ cat /etc/debian_version
  => 10.4

:~ $ cat /etc/os-release
    => PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
          NAME="Raspbian GNU/Linux"
          VERSION_ID="10"
          VERSION="10 (buster)"
          VERSION_CODENAME=buster
          ID=raspbian
          ID_LIKE=debian
          HOME_URL="http://www.raspbian.org/"
          SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
          BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

I used preconfigured Raspberry Pi OS from navio->
Emlid Raspberry Pi OS Buster [XZ, 662 MB], (md5)
link-> https://docs.emlid.com/navio2/configuring-raspberry-pi
link-> https://files.emlid.com/images/emlid-raspbian-20200922.img.xz

for compiler --->
I used pre-installed compiler. I also tried gcc-arm-none-eabi-10.3-2021.10-aarch64-linux.tar.bz2

Linux AArch64 Tarball
MD5:3fe3d8bb693bd0a6e4615b6569443d0d

nothing worked. 
Thanks!

<h2 style="box-sizing: border-box; color: rgb(28, 30, 33); font-family: GothamProMedium; font-weight: 500; line-height: 42px; margin-top: calc(var(--ifm-h2-vertical-rhythm-top)*var(--ifm-leading)); margin-right: ; margin-bottom: calc(var(--ifm-heading-vertical-rhythm-bottom)*var(--ifm-leading)); margin-left: ; font-size: 1.875rem !important; letter-spacing: -0.04em; --ifm-h2-font-size:2rem; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><a aria-hidden="true" tabindex="-1" class="anchor anchor__h2 anchorWithStickyNavbar_31ik" id="write-image-to-sd-card" style="box-sizing: border-box; color: var(--ifm-link-color); text-decoration: none; transition: color var(--ifm-transition-fast) var(--ifm-transition-timing-default); display: block; position: relative; top: -0.5rem; scroll-margin-top: calc(var(--ifm-navbar-height) + 0.5rem);"></a></h2><br class="Apple-interchange-newline">

sandeshneupane, 2021-11-19:
link for above tar- https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads