PX4/PX4-Autopilot #18651: loop of mavftp protocol crash drone

BOB4Drone, 2021-11-16:
**Describe the bug**
sending more then 1400 packet at px4 firmware, crash nuttx OS and reboot device

**To Reproduce**
set opcode with 9 (create directory) and set payload with random data ( more then 105 byte). if sum of sent data is more then 1400 packet, it will crash nuttX OS and reboot

**Expected behavior**
Crash OS and reboot device

**Drone (please complete the following information):**
- all nuttX based drone

**Additional context**
- i generated random directory packet with Python and sent generated packet to px4 mini (px4_fmu_v5-default) with Serial
- we need only 2 seconds to Crash drone
- i think it becomes from lack of ram ( nuttX fs is vfs, so it makes filesystem on ram). i think there should be limited directory length on NuttX

bkueng, 2021-11-16:
Good find, can you provide the script?

BOB4Drone, 2021-11-16:
https://github.com/BOB4Drone/Project_4Drone/blob/main/fuzzer/mavFTPfuzzer.py

here is my poc code with python

bkueng, 2021-11-16:
Thanks @BOB4Drone. You are right, it fills up the memory with random files.
I'm adding a path restriction in https://github.com/PX4/PX4-Autopilot/pull/18655

closed 2021-11-16 by dagar