PX4/PX4-Autopilot #18658: Preflight Fail: High Accelerometer Bias in simulation

codingCoffee, 2021-11-17:
**Describe the bug**
Trying to arm the drone raises preflight fail errors in gazebo simulation.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `make px4_sitl_default gazebo`. This should launch gazebo.
2. `roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`
3. Open QGC
4. Arm Drone in QGC, in Position Mode
5. See error

**Expected behavior**
The drone is supposed to get armed, the propellers are suppose to rotate, and we should see a message stating the drone is ready to fly. 

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=a0cb5c7b-62ae-435d-b793-aec8dedf8fac

Add screenshots to help explain your problem.
[image]

**Drone (please complete the following information):**
- This is iris drone in gazebo simulation

**Additional context**
I haven't been able to locate the reason for the error. After setting up the system, I was able to fly the drone in mission mode inside the simulation without any issues using the command below, however after 3-4 flights this error suddenly appeared and I haven't been able to resolve it since. 
```
python3 mission_test.py MC_mission_box.plan
```

dagar, 2021-11-17:
Try clearing your acceleration calibration parameters.

```
CAL_ACC0_ID, 1310988
CAL_ACC0_PRIO, 50
CAL_ACC0_ROT, -1
CAL_ACC0_TEMP, -1000.0
CAL_ACC0_XOFF, -0.258414089679718
CAL_ACC0_XSCALE, 1.0
CAL_ACC0_YOFF, -0.06120181828737259
CAL_ACC0_YSCALE, 1.0
CAL_ACC0_ZOFF, 0.5727328658103943
CAL_ACC0_ZSCALE, 1.0
CAL_ACC1_ID, 1310996
CAL_ACC1_PRIO, 50
CAL_ACC1_ROT, -1
CAL_ACC1_TEMP, -1000.0
CAL_ACC1_XOFF, -0.262330025434494
CAL_ACC1_XSCALE, 1.0
CAL_ACC1_YOFF, 0.05466032028198242
CAL_ACC1_YSCALE, 1.0
CAL_ACC1_ZOFF, 0.3812338411808014
CAL_ACC1_ZSCALE, 1.0
CAL_ACC2_ID, 1311004
CAL_ACC2_PRIO, 50
CAL_ACC2_ROT, -1
CAL_ACC2_TEMP, -1000.0
CAL_ACC2_XOFF, -0.2617260813713074
CAL_ACC2_XSCALE, 1.0
CAL_ACC2_YOFF, 0.05408119782805443
CAL_ACC2_YSCALE, 1.0
CAL_ACC2_ZOFF, 0.3815549612045288
CAL_ACC2_ZSCALE, 1.0
CAL_ACC3_ID, 0
```

There's a new feature where the EKF learned accelerometer bias is saved back to the calibration parameters incrementally after each flight. If this is what you're getting after a few normal SITL flights it appears we have a bug somewhere.

codingCoffee, 2021-11-17:
@dagar thank you so much for the help!

TL;DR
Clearing acceleration calibration parameters accompanied by recalibration should probably fix the issue in the real world, however given this problem was occurring in HIL simulation, recalibration was not so easy since it entails holding the quadcopter in listed positions inside the simulator. 
A quick fix I found was directly resetting the EKF values, by using the commands `ekf2 stop` and `ekf2 start` in pxh console.