PX4/PX4-Autopilot #18660: CUAV F9P Preflight Fail with v1.12.3

YencyAllegro, 2021-11-17:
**Describe the bug**
Trying to flight with two GPS connected, First GPS (Pixhawk 4 GPS Module, Holybro M8N) connected on GPS1 port and Secondary GPS (CUAV F9P) connected on GPS2 port. 
When only GPS1 is plugged, everything works fine. The drone can arm, takeoff, switch to Stabilized, Hold and Position modes, and land
PROBLEM WITH TWO GPS : When the secondary GPS is plugged, its parameters set (GPS_2_CONFIG : GPS2, GPS_2_GNSS : 29, GPS_2_PROTOCOL : u-blox, SER_GPS2_BAUD : Auto) and the compass recalibrated it show various “Preflight” and “Preflight Fail”

**To Reproduce**
Steps to reproduce the behavior:
    1. Pixhawk 5X (Holybro) mounted
    2. Firmware : px4_fmu-v5x_default.px4
    3. Version v.1.12.3
    4. First GPS (Pixhawk 4 GPS Module, Holybro M8N) connected on GPS1 port, Secondary GPS (CUAV F9P) connected on GPS2 port.
    5. Open QGC 
    6. Recalibrate compass
    7. Try to arm

**Expected behavior**
Be able to Takeoff, Flight en Stabilized, Position and Hold Modes and Land, in v1.12.3 with two GPS plugged. Have the data of both GPS collected (in plotJuggler3 : see “vehicle_gps_pos00” and “vehicle_gps_pos01” in the Timeseries list.

**Additional context**
This issue didn’t occur in v.1.10.1. When switching back to v1.10.1, both GPS are taken into account and both parameters “vehicle_gps_pos00” and “vehicle_gps_pos01” are visible on plotJuggler3.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=b42b4881-7535-47ec-8f1b-8402c94a5622 

Add screenshots to help explain your problem.
With only one GPS : IMAGE
[image]

With both GPS : 3 IMAGES
[image]

[image]

[image]

**Drone**
- Describe the type of drone : Quadrocopter

dagar, 2021-11-19:
> Have the data of both GPS collected (in plotJuggler3 : see “vehicle_gps_pos00” and “vehicle_gps_pos01” in the Timeseries list.

In newer PX4 each GPS data source is published  as sensor_gpsX, then it's aggregated by the sensors module and either a primary is selected and published as `vehicle_gps_position` or multiple sources are blended.

With v1.12.3 or master will all hardware connected can you enable logging from boot (SDLOG_MODE) and add "Sensor comparison" and "Estimator replay (EKF2)". If you then do a full mag calibration and try to arm that log should show us everything.

Katawann, 2021-11-19:
> With v1.12.3 or master will all hardware connected can you enable logging from boot (SDLOG_MODE) and add "Sensor comparison" and "Estimator replay (EKF2)". If you then do a full mag calibration and try to arm that log should show us everything

ok we do that and come back with a new log

YencyAllegro, 2021-11-22:
Hi, I have tried again  adding the parameters you asked for. SDLOG_MODE = "from boot until shutdown" and in SDLOG_PROFILE I have "Default set", "Estimator replay (EKF2)", "High rate" and "Sensor comparison" that are checked.

Here is the resulting log : https://review.px4.io/plot_app?log=c3f3e17a-ee43-4a99-aa28-5d3c7849281f 

I didin't changed anything else, but this time I was able to takeoff and land and switch to hold and position modes. But I want to be sure that the second data is parametred correctly and its data blended with the data of the first GPS.
(/!\ I work without the propellers because I already crashed the drone some times and I want to be sure everything work fine before flying "for real".)

YencyAllegro, 2021-11-22:
I will close the issue here, and I will maybe reopen it if the problem reappears in the future.

closed 2021-11-22 by YencyAllegro