PX4/PX4-Autopilot #18670: [ROS2, px4_sitl_rtps] position_setpoint_triplet doesn't work in offboard control mode

JohnnyLiu2886, 2021-11-18:
Hi everyone, I am working on an autonomous boat using pixhawk 4 and jetson nano. It is using Ubuntu 18.04. I am going to control the boat using ROS2-dashing and PX4. 

I was successful to use trajectory_setpoint in "make px4_sitl_rtps gazebo_boat". It can arm, switch to offboard mode and reach the local setpoint.

However, I want to decide the setpoint for the boat using GPS position in WGS84 instead of NED frame. Therefore, I tried to use uorb topic "position_setpoint" but the boat had no response after arming and switching to offboard mode. I made sure that the simulator can receive and send gps through uorb topic "Vehicle global position". Hence, the boat should be able to receive gps position "lon and lat".

After that, I tried "position_setpoint_triplet" with current and next position. The current position is located using the data from uorb topic "Vehicle global position". However, it still gives no response to me even it armed, took off and switched to offboard mode successfully.

1. Is there some important parameter I did not set for the uorb message "position_setpoint"?
2. Is global position of "position_setpoint" worked in offboard mode?
[image]

96moustafa, 2021-12-08:
AFAIK, position setpoint only works with NED setpoints and not with global ones. What you can do is to calculate the NED offset between your current global position and the goal position then update the position_setpoint accordingly.

JohnnyLiu2886, 2021-12-10:
> AFAIK, position setpoint only works with NED setpoints and not with global ones. What you can do is to calculate the NED offset between your current global position and the goal position then update the position_setpoint accordingly.

Thanks for your response. I gave up using gps position setpoint directly half month ago.

closed 2021-12-10 by JohnnyLiu2886