PX4/PX4-Autopilot #18672: Longer logging ends in error of FAT32 filesystem, Corrupted SD card

roman-dvorak, 2021-11-18:
Hi, we repeatedly have a problem with logging to the SD card. It happened to us several times that the card was corrupted during the inflight logging. The SD card or filesystem during read by computer seems to be flagged as read-only. And the log file was not complete. 

For the first time, it happened to us on 7. 10. 2021 with takeoff mode (#18582), which was based on a few days old PX4  master version. At that time, I thought it was due to a large number of debug text logs (`PX4_INFO("...")`). In the meantime, we tried to use a new (32GB SD card)[], format the card (completely erased) to FAT32. And the problem was repeated several times. Therefore we didn't have a log that we really needed.. It does not matter that we create a smaller 4GB FAT32 volume on SDcard. The situation was almost the same, corrupted filesystem with missing logs and sometimes other files which were previously on SDcard. 

Next time, I limited number of `PX4_INFO` messages to the minimum number of these logs. And the SD card was corrupted again. With such FW. But it was the  Firmware with my modifications. 

The first used card was:
[image]

After a debate in (#18419), we decided to test this “airspeed mode” PR with our second autogyro (FMU v5, CUAV NANO) and this was almost unmodified FW (f814a52). But during the flight, the SDcard filesystem corruption happened again. This piece of autogyro (the same autopilot, the same SDcard, ..) was previously used in the same configuration with several months old master FW. Without any signs of this problem. 

This always happened after a few seconds of flight. I don’t see any clear cause for this behaviour in the same fragments of the logs, retrieved from SDcard after fsck recovery,  containing the takeoff 

[image]
[image]

I tried to repeat this on the ground. But I was unsuccessful. Logging worked for a very long time. 

From these experiments, I conclude that this is a problem in Firmware (I don’t know at what level). 

During the first attempts (when this issue occurred) I used the internal reader of my laptop. During the next experiments, I used an external uSD card reader from [Kingstone](https://www.kingston.com/czech/en/memory-card-readers/usb-3-0-high-speed-media-reader). But the result was the same.

dagar, 2021-11-18:
Thanks for the detailed report. We're currently investigating this as a blocking issue for the next release (v1.13).

roman-dvorak, 2021-11-21:
Hi, I've been trying to make more attempts to get to the state where you can log.

1. I turned on the complete UAV assembly. Pressed safety switch and I forced arm through QGC command line. 
2. Logging on (confirmed by changing log file size)
3. I waited a while and the logfile stopped growing. It took only several minutes. 

Subsequently, in nuttx dmesg was: 
[image]

In this state, it is not possible to restart logger: 
[image]
It writes that it only records the mission log. 

I was able to repeat this several times (4x). Some logs could be uploaded to review.px4.io and some not (without any error message). However, all could be displayed in the plotjuggler. Some I was unable to get due to a corrupted SD card. 

log files, that ended "in-flight" (in armd state). 
[15_22_37.zip](https://github.com/PX4/PX4-Autopilot/files/7576591/15_22_37.zip)
[15_41_42.zip](https://github.com/PX4/PX4-Autopilot/files/7576592/15_41_42.zip)

Top (in time of corrupted logger):
[image]
[image]

There is an high CPU usage. By `log_writer_file`. 

Sometimes I was not able to connect the card to the computer (ubuntu 20.04). With these errors:
[image]
[image]

Tested with these cards:
[image]

[image]
[image]

dagar, 2021-11-21:
@roman-dvorak we may have a fix, could you please try https://github.com/PX4/PX4-Autopilot/pull/18687?

To be paranoid and convince myself the card and file system is in good shape I've been exhaustively testing all blocks. https://gist.github.com/dagar/8d002760ee1f8e76d1bd88c16a469d24#file-linux_sdcard_test_and_format-sh-L8

roman-dvorak, 2021-11-23:
Booth SD cards look well. It passed the test without any error. Without any bad sector. 

SanDisc Ultra 16 GB:
<pre>
Object /org/freedesktop/UDisks2/block_devices/sdf is not a mountable filesystem.

*****  BADBLOCK *****

Hledají se špatné bloky v režimu čtení i zápis
Od bloku 0 do 15558143
Zkouším s náhodným vzorkem: hotovo                                                 
Čtení a porovnání: hotovo                                                 
Průchod dokončen, nalezeno 0 špatných bloků (0/0/0 chyb).

6.42user 15.65system 25:18.85elapsed 1%CPU (0avgtext+0avgdata 2672maxresident)k
31118848inputs+31116288outputs (12major+147minor)pagefaults 0swaps

*****  F3PROBE *****

F3 probe 7.2
Copyright (C) 2010 Digirati Internet LTDA. This is free software; see the source for copying conditions.

WARNING: Probing normally takes from a few seconds to 15 minutes, but it can take longer. Please be patient.

Good news: The device `/dev/sdf&apos; is the real thing

Device geometry:
	         *Usable* size: 14.84 GB (31116288 blocks)
	        Announced size: 14.84 GB (31116288 blocks)
	                Module: 16.00 GB (2^34 Bytes)
	Approximate cache size: 0.00 Byte (0 blocks), need-reset=no
	   Physical block size: 512.00 Byte (2^9 Bytes)

Probe time: 2&apos;29&quot;
 Operation: total time / count = avg time
      Read: 829.0ms / 4814 = 172us
     Write: 2&apos;27&quot; / 4192321 = 35us
     Reset: 0us / 1 = 0us

0.95user 0.67system 2:29.60elapsed 1%CPU (0avgtext+0avgdata 3404maxresident)k
5382inputs+4192321outputs (3major+362minor)pagefaults 0swaps
</pre>

and the second SD (SanDisc Ultra 32 GB):

<pre>
*****  BADBLOCK *****

Hledají se špatné bloky v režimu čtení i zápis
Od bloku 0 do 31166975
Zkouším s náhodným vzorkem: hotovo                                                 
Čtení a porovnání: hotovo                                                 
Průchod dokončen, nalezeno 0 špatných bloků (0/0/0 chyb).
8.24user 19.18system 32:30.84elapsed 1%CPU (0avgtext+0avgdata 3052maxresident)k
62334736inputs+62333952outputs (6major+157minor)pagefaults 0swaps

*****  F3PROBE *****

F3 probe 7.2
Copyright (C) 2010 Digirati Internet LTDA.
This is free software; see the source for copying conditions.

WARNING: Probing normally takes from a few seconds to 15 minutes, but
         it can take longer. Please be patient.

Good news: The device `/dev/sdf' is the real thing

Device geometry:
	         *Usable* size: 29.72 GB (62333952 blocks)
	        Announced size: 29.72 GB (62333952 blocks)
[... 72 lines cut ...]
iteration 2 took 2271921 us: OK
iteration 3 took 2345547 us: OK
iteration 4 took 2330682 us: OK
nsh&gt; 
nsh&gt; sd_bench -u -v
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  2032.37 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  1:  1874.72 KB/s, max write time: 67 ms (=  59.70 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  2:  2014.84 KB/s, max write time: 49 ms (=  81.63 KB/s), fsync: 12 ms
INFO  [sd_bench]   Run  3:  1846.66 KB/s, max write time: 67 ms (=  59.70 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run  4:  1964.14 KB/s, max write time: 47 ms (=  85.11 KB/s), fsync: 54 ms
INFO  [sd_bench]   Avg   :  1946.65 KB/s
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Read Speed of 4913 blocks
INFO  [sd_bench]   Run  0:  2970.85 KB/s, max read/verify time: 7 ms (= 571.43 KB/s)
INFO  [sd_bench]   Run  1:  2891.55 KB/s, max read/verify time: 5 ms (= 800.00 KB/s)
INFO  [sd_bench]   Run  2:  2962.37 KB/s, max read/verify time: 5 ms (= 800.00 KB/s)
INFO  [sd_bench]   Run  3:  3083.00 KB/s, max read/verify time: 2 ms (=2000.00 KB/s)
INFO  [sd_bench]   Avg   :  2955.36 KB/s 4913 blocks read and verified
nsh&gt; 
</pre>

I merged your PR (#18687) into autogyro takeoff PR (#18582), which resulted into a commit: https://github.com/ThunderFly-aerospace/PX4Firmware/commit/252e4e94f6e07d8d155a628cd67d1b08c8525398. Everything looks good during these bench tests. But I can't confirm it, right now. I tried to do the same operations as in the previous attemt. And with each attempt, a complete log was created. In order to confirm the functionality of these improvements, we must do some flight-test in real enviroment.

closed 2021-11-25 by dagar

roman-dvorak, 2021-12-04:
Thanks! I can confirm that this problem did not occur during the flight test.