PX4/PX4-Autopilot #18679: OFFBOARD CONTROL USING MAVROS

bhaskar-glitch, 2021-11-19:
Hi,
I’m using **pixhawk 2.4.8**+**jetson xavier** and i’m using VINS_FUSION for locatization of my drone…now please help that how i can use **OFFBOARD** mode and what topic i've to publish to make **my drone fly** and also my drone should take inputs of local position from **VINS_FUSION** because i'm not using **GPS**...
Please help…i’m done with all the things but stuck in flying part of drone…?

Jaeyoung-Lim, 2022-01-10:
@bhaskar-glitch Please be more specific on which aspect you need help. Otherwise offboard mode is reasonably well documented in http://docs.px4.io/master/en/flight_modes/offboard.html and http://docs.px4.io/master/en/ros/mavros_offboard.html

bhaskar-glitch, 2022-01-10:
> @bhaskar-glitch Please be more specific on which aspect you need help. Otherwise offboard mode is reasonably well documented in http://docs.px4.io/master/en/flight_modes/offboard.html and http://docs.px4.io/master/en/ros/mavros_offboard.html

Thank you for your reply bta I've solved this issue so far..

Jaeyoung-Lim, 2022-01-10:
@bhaskar-glitch Thanks for the update and glad to hear that it worked out. Closing the issue

closed 2022-01-10 by Jaeyoung-Lim