PX4/PX4-Autopilot #18683: H7/F7 UAVCAN bootloader implem

TregliaRaphael, 2021-11-20:
Hi, 
I discovered uavcan bootloader for some of the board like can_gps_v1 for cuav and some others but i don't find any can bootloader for stm32H7 and F7.
Is canbootloader exist for the H7/F7 target or is it still pending?
There is all arch available for stm32 here but not h7 for now [Can bootloader arch stm](https://github.com/PX4/PX4-Autopilot/tree/master/platforms/nuttx/src/canbootloader/arch/stm) 

Did you know if this was started Somewhere because i could help?
Moreover, i dont find any specific canbootloader f7 build in makefile could you help me to find one?

I'm actually creating a fc which could match with canbootloader f7 or h7.

Thanks you !

dagar, 2021-11-20:
STM32F7 is fully working, see boards/freefly/can-rtk-gps as an example. I haven't needed to look into STM32H7 yet, but it shouldn't be hard to extend.

TregliaRaphael, 2021-11-20:
Thanks a lot :)
I'll check in this way

TregliaRaphael, 2021-11-20:
Could you just help me to identify the parts I need to do to achieve the H7 uavcan bootloader ? That's the first time I'm doing these kinds of things so I don't want to forget anything if possible.
Should I just do the H7 arch of the uavcan bootloader? in this folder https://github.com/PX4/PX4-Autopilot/tree/master/platforms/nuttx/src/canbootloader/arch/stm

dagar, 2021-11-20:
> Could you just help me to identify the parts I need to do to achieve the H7 uavcan bootloader ? That's the first time I'm doing these kinds of things so I don't want to forget anything if possible. [...]

You could start by taking a look at what was added for F7 support along with the freefly rtf board. https://github.com/PX4/PX4-Autopilot/pull/17553

The main difference that comes to mind is that H7 will have a different minimal CAN driver. https://github.com/PX4/PX4-Autopilot/blob/70d5377aa7815149fc799630cc3bb7a97f4cd762/platforms/nuttx/src/canbootloader/arch/stm/stm32f7/drivers/can/driver.c

TregliaRaphael, 2021-11-22:
When you say different minimal CAN, do you just mean adapting clock speed/parameters and addresses ?

dagar, 2021-11-22:
> When you say different minimal CAN, do you just mean adapting clock speed/parameters and stm addresses ?

The CAN driver for stm32h7 is a bit different than stm32f4/f7 (bxcan). This is the minimal CAN driver in the bootloader for stm32f7. https://github.com/PX4/PX4-Autopilot/blob/70d5377aa7815149fc799630cc3bb7a97f4cd762/platforms/nuttx/src/canbootloader/arch/stm/stm32f7/drivers/can/driver.c

You could take a look at the full fledged UAVCAN drivers to see the differences between them. https://github.com/PX4/PX4-Autopilot/tree/70d5377aa7815149fc799630cc3bb7a97f4cd762/src/drivers/uavcan/uavcan_drivers

TregliaRaphael, 2022-01-10:
Hi,
I had the time to work on it since the last time we discuss, and I'll like to know how can I share you my work through branches / pushes on the px4 github?

sanyanosnikov, 2022-11-08:
@TregliaRaphael Hi. If you have already made a canbootloader, could you share it? Thanks

TregliaRaphael, 2022-11-20:
Hi,
I'hv done my proper can bootloader outside of the PX4 ecosystem because i prefer to control everything i'm doing. I won't share it sorry.
There was too much work for me as a newby on Nuttx because the H7 is not fully implemented on the driver side or the difference between F7 and H7 were too different for me to take the risk to modify important things.

I had a can bootloader branch that compile but there still was a non negligeable job to do to make it works. It's just an hour work.

hamishwillee, 2025-10-29:
Closing - more than just a little stale. If this is still a problem, please replicate on current main or stable and repost.

closed 2025-10-29 by hamishwillee