PX4/PX4-Autopilot #18702: Autopilot does not control airspeed during decent (VTOL) v1.12.3

moreba1, 2021-11-23:
Hi

 In my sitl test in v1.12.3 , during decent in loitering TECS control loop does not control airspeed. pitch setpoint and throttle did not react for controling airspeed.
https://review.px4.io/plot_app?log=973ca5c7-0529-4248-85f2-e7bd8430b981

[image]

[image]

[image]

sfuhrer, 2021-11-26:
Doesn't look crazy off to me, the SITL models are not perfectly well tuned, and even on a reasonable well tuned vehicle airspeed errors of this range of 0.5-1m/s is normally acceptable. 
Nevertheless, with https://github.com/PX4/PX4-Autopilot/pull/18729 the tuning should be slightly improved. 

If you want to play around the tuning yourself: If you want to control the airspeed closer then you can reduce FW_T_TAS_TC, reduce FW_T_TAS_R_TC and increase FW_T_SPDWEIGHT. Usually giving more priority to airspeed tracking leads to slightly worse altitude tracking.

moreba1, 2021-11-27:
I know that TECS control between airspeed and altitude with FW_T_SPDWEIGHT but TECS must never let that airspeed increases more than FW_AIRSPD_MAX even with FW_T_SPDWEIGHT=1 and bad tuned sink rate parameters (high sink rate). 
I think that if user tuned sink rate setpoint wrong , autopilot must prefer FW_AIRSPD_MAX over FW_T_SINK_R_SP , FW_T_SINK_MAX and prevent overspeed condition.

In this second SITL test during descent airspeed reaches 30 m/s but FW_AIRSPD_MAX is 19 m/s.
FW_T_SINK_R_SP = 7
FW_T_SINK_MAX = 8
FW_AIRSPD_MIN=2.2 
FW_AIRSPD_MAX= 19 

https://review.px4.io/plot_app?log=6dc68a08-ebf2-404c-b7ad-cd38d6661047

hamishwillee, 2025-10-29:
Closing - more than just a little stale. If this is still a problem, please replicate on current main or stable and repost.