PX4/PX4-Autopilot #18706: Fixed-wing landing abort loiter altitude behavior inconsistent with comment

thomasteisberg, 2021-11-23:
When a fixed-wing landing is aborted, the vehicle switch to hold mode with a loiter set over the original landing setpoint. The logic determining the altitude of this hold (current master branch) is:

```
// loiter at the larger of MIS_LTRMIN_ALT above the landing point
//  or 2 * FW_CLMBOUT_DIFF above the current altitude
const float alt_landing = get_absolute_altitude_for_item(_mission_item);
const float alt_sp = math::max(alt_landing + _navigator->get_loiter_min_alt(),
			       _navigator->get_global_position()->alt + 20.0f);
```

The comment and the code inconsistent here. The PR discussion does not reference the change, so I'm wondering if it's possible the change in behavior was made by mistake.

Commit where the behavior first changed: 956935141e72d8252ef76814147434c5fa7fa715
Associated PR: https://github.com/PX4/PX4-Autopilot/pull/8497

I believe neither behavior is actually documented outside of the code. I'd be happy to help improve the documentation if other can weigh in on what they feel the intended behavior is supposed to be and why.

ryanjAA, 2022-09-20:
Is this still the case?

sfuhrer, 2023-02-06:
This has been resolved (now does it at least at NAV_MIN_LTR_ALT, or current altitude).

closed 2023-02-06 by sfuhrer