PX4/PX4-Autopilot #18713: Quaternion conversion does not work for pitch -90 degrees

dayjaby, 2021-11-24:
**Describe the bug**

```
matrix::Quatf q_comp(matrix::Eulerf(0, math::radians(-89.9f), 0));
```

gives approximately
```
0.707, 0.00, -0.707, 0.00
```

However
```
matrix::Quatf q_comp(matrix::Eulerf(0, math::radians(-90.0f), 0));
```

gives
```
1.00, 0.00, 0.00, 0.00
```

Trying to use that to compare a gimbal_device_attitude_status to a certain setpoint; that's where I noticed this bug.

dayjaby, 2021-11-24:
I can give it a try to fix it. But zero prior experience with dcm matrices.

bresch, 2021-11-30:
@dayjaby Which version of https://github.com/PX4/PX4-Matrix are you using? I just tried it and I get the correct result (`q = [0.707, 0.00, -0.707, 0.00]`) when at pitch = 90deg

dayjaby, 2021-12-11:
@bresch You are right. There was something wrong with my test. And I just realized we already have unit tests for these cases: https://github.com/PX4/PX4-Autopilot/blob/master/src/lib/matrix/test/attitude.cpp

closed 2021-12-11 by dayjaby