PX4/PX4-Autopilot #18715: Get bootloader version/hash/code in PX4

RicardoM17, 2021-11-24:
Hey all,

I was wondering if there's currently any way to check the current Bootloader version (either git tag/hash or even some other code that could be matched with externally) inside PX4. I need this all the way on the ROS side but obviously I would already be more than happy to get it from within PX4.

Thanks in advance

dagar, 2021-11-26:
There's a major bootloader version you can read over USB or serial as the system is booting, but there isn't much granularity. https://github.com/PX4/PX4-Bootloader/blob/d487c4c6f5e150f527403f11700549f0d41d94db/bl.c#L83

RicardoM17, 2021-11-26:
Thanks for the reply @dagar. That should do the trick :)

closed 2021-11-26 by RicardoM17