PX4/PX4-Autopilot #18717: Trying to get in touch regarding a security issue

zidingz, 2021-11-25:
Hey there!

I belong to an open source security research community, and a member (@dwisiswant0) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

dagar, 2021-11-26:
As a start could you join us on PX4 slack and message me (@dagar)? https://github.com/PX4/PX4-Autopilot#readme

dwisiswant0, 2022-03-26:
It's been 120+ days and looks like the report hasn't reached you yet, you should be able to see the submission directly at https://www.huntr.dev/bounties/6f8a0616-9587-4c1e-817b-a3977566bd46/, @dagar.

dagar, 2022-03-27:
I don't think this is a concern, this regular expression is only used at compile time for PX4 parameters (added by developers int tree).
[image]