PX4/PX4-Autopilot #18718: ALT broken in the VTOL RTL

Kim-zhi-jiang, 2021-11-25:
The altitude keeps falling at RTL to home.

RomanBapst, 2021-11-25:
@Kim-zhi-jiang Please add a log or a better explanation.

Kim-zhi-jiang, 2021-11-25:
@RomanBapst   [log](https://logs.px4.io/plot_app?log=23730b9e-596b-4cff-980e-178ff4b928b6)

RomanBapst, 2021-11-25:
@Kim-zhi-jiang Can you explain how you reproduced this? Any code changes? What was your command for the simulation?
Did you change params?

Kim-zhi-jiang, 2021-11-25:
@RomanBapst 
Not Changes code,
PX4_SIM_SPEED_FACTOR=3 PX4_HOME_LAT=22.5692507 PX4_HOME_LON=113.4770233 PX4_HOME_ALT=29.5 make -j4 px4_sitl gazebo_standard_vtol
I don't know where it happened Maybe commander.cpp,
Test Master everyday.

closed 2021-11-26 by Kim-zhi-jiang

NicolasM0, 2021-12-16:
It seems that your RTL point altitude is set to 0m so 29m below your ground altitude.

Kim-zhi-jiang, 2021-12-16:
@NicolasM0  Can be easily reproduced,  git clone master  make px4_sitl gazebo_standard_vtol
RTL then...

Kim-zhi-jiang, 2021-12-16:
@NicolasM0   Maybe from here #18395