PX4/PX4-Autopilot #18732: Brushless motor RPM driver

jmailloux, 2021-11-26:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or Slack (you can find an invite link on this project README).

**Describe problem solved by the proposed feature**
I am using the PX4 software to build a hydrofoiling electric boat.  One of the things I needed was to get an accurate reading of how fast the motor was spinning.  To do this, I used this sensor: https://www.offshoreelectrics.com/proddetail.php?prod=ose-83279-011.  To get the sensor to work, I wrote a driver for it using the pwm_input driver.

**Describe your preferred solution**
I have a driver written for this part: https://www.offshoreelectrics.com/proddetail.php?prod=ose-83279-011.  I'd like to contribute it back to the main code base.  Can submit a pull request if you think it is worth it.

**Describe possible alternatives**
I looked at various other RPM sensors - optical ones, ones integrated into the motor controller.  Settled on this hardware because it is cheap, and can be used for any brushless DC motor / motor controller combination.  This part is quite accurate - I compared the readings to what an optical sensor read, and they were essentially the same.

**Additional context**

ryanjAA, 2021-12-01:
Cool. Do you have a pr with your code so someone can review it? That would be the next step.

jmailloux, 2021-12-03:
Sounds good.  I will post a PR next week.

jmailloux, 2021-12-06:
OK.  PR ready.  Wasn't really sure what to call the driver.  You will probably have a better suggestion.