PX4/PX4-Autopilot #18738: ninja: fatal: invalid -j parameter

boonflies, 2021-11-28:
After cloning,
**git clone https://github.com/PX4/Firmware.git
**
and running,
no_sim=1 make px4_sitl gazebo

I am receiving the following error:
-- Found GStreamer: adding gst_camera_plugin
-- Found GStreamer: adding gst_video_stream_widget
-- Configuring done
-- Generating done
-- Build files have been written to: /home/boon/src/Firmware/build/px4_sitl_default/build_gazebo
[650/833] Performing build step for 'sitl_gazebo'
ninja: fatal: invalid -j parameter
[657/833] Building CXX object src/modu...odules__mavlink.dir/mavlink_main.cpp.o
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/boon/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/boon/src/Firmware/build/px4_sitl_default/build_gazebo -- -j 0
[663/833] Building CXX object src/modu...odules__mavlink.dir/mavlink_main.cpp.o
ninja: build stopped: subcommand failed.
Makefile:230: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

dagar, 2021-11-28:
What's your development environment? Linux distribution, version, etc

boonflies, 2021-11-28:
I was running on Ubuntu 18.04 . When I downloaded the last stable version and tried to build px4, did not get any error.

dagar, 2021-11-28:
> I was running on Ubuntu 18.04 . When I downloaded the last stable version and tried to build px4, did not get any error.

Can you try removing the `-j` here? https://github.com/PX4/PX4-Autopilot/blob/2870b5bbb167d0571d22790d4c350600fbffe684/platforms/posix/cmake/sitl_target.cmake#L70

We might need to add a ninja-build version check.

mollaemin, 2021-12-18:
> After cloning, **git clone https://github.com/PX4/Firmware.git ** and running, no_sim=1 make px4_sitl gazebo
>
> I am receiving the following error: -- Found GStreamer: adding gst_camera_plugin -- [...]

same problem no solution on all web

julianoes, 2021-12-20:
@mollaemin try the suggestion above and remove `-j ` in https://github.com/PX4/PX4-Autopilot/blob/2870b5bbb167d0571d22790d4c350600fbffe684/platforms/posix/cmake/sitl_target.cmake#L70.

Also, please paste the output of:
```
ninja --version
```
And:
```
cmake --version
```

julianoes, 2021-12-20:
I just tested this on my Ubuntu 18.04.6 and it builds fine:

```
ninja --version
1.8.2
cmake --version
cmake version 3.10.2
```

julianoes, 2021-12-20:
Oh, it looks like the argument `0` is not allowed for Ninja.

Could you change `DEBUG` to `FATAL_ERROR` here:
https://github.com/PX4/PX4-Autopilot/blob/2870b5bbb167d0571d22790d4c350600fbffe684/platforms/posix/cmake/sitl_target.cmake#L52-L53

and then paste the output here?

mollaemin, 2021-12-22:
> Oh, it looks like the argument `0` is not allowed for Ninja.
> 
> Could you change `DEBUG` to `FATAL_ERROR` here:
> 
> https://github.com/PX4/PX4-Autopilot/blob/2870b5bbb167d0571d22790d4c350600fbffe684/platforms/posix/cmake/sitl_target.cmake#L52-L53
> 
> and then paste the output here?

Thanks for the answer. I was using VM for Ubuntu. After various problems, I decided to install ubuntu on my local drive and it works fine now. It was probably about VM and available cores

julianoes, 2021-12-22:
Ok, it's good that you have solved your problem but it's suboptimal because now we don't have any data to fix this :smirk:.

If anyone else has this problem, please paste the output as I mentioned it above.

JamesHoi, 2022-01-16:
Hi there. I am having the same issue.
Env: Ubuntu 18.04 on VMware

> Oh, it looks like the argument `0` is not allowed for Ninja.
> 
> Could you change `DEBUG` to `FATAL_ERROR` here:
> 
> https://github.com/PX4/PX4-Autopilot/blob/2870b5bbb167d0571d22790d4c350600fbffe684/platforms/posix/cmake/sitl_target.cmake#L52-L53
> 
> and then paste the output here?

After I change `DEBUG` to `FATAL_ERROR`, I got the output:

-- PX4 version: v1.13.0-alpha1-4065-g0ce44cebb7
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- PX4 config file: /home/james/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ROMFS: ROMFS/px4fmu_common
CMake Error at platforms/posix/cmake/sitl_target.cmake:52 (message):
  2 logical cores detected and 1372 megabytes of memory available.

  Limiting sitl_gazebo and simulation-ignition concurrent jobs to 0
Call Stack (most recent call first):
  platforms/posix/CMakeLists.txt:121 (include)

-- Configuring incomplete, errors occurred!
See also "/home/james/PX4-Autopilot/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
Error: /home/james/PX4-Autopilot/build/px4_sitl_default is not a directory
Makefile:235: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

mollaemin, 2022-01-16:
I still don't know what is the problem but while we were trying to solve it my friend told me about it might be about the cores available to use in VM so we decided to change 
```
BUILD_COMMAND ${CMAKE_COMMAND} --build <BINARY_DIR> -- -j ${parallel_jobs} 
```
${parallel_jobs} parameter to 0 or 1 "not sure which one"  and it worked without a problem however it was quite slow. Maybe this info can help the @julianoes to solve the problem

PR #19018 (2022-01-17): cmake: use at least 1 core to build simulator
If we use -j 0, ninja fails with invalid argument.

Closes #18738.
closed 2022-01-17 by dagar