PX4/PX4-Autopilot #18745: Integration of New INS and Air Data Sensors

flybrianfly, 2021-11-30:
Hello, my company (https://bolderflight.com/) is developing new Inertial Navigation System (INS / IMU + GNSS + EKF) and air data (pressure, speed, altitude, OAT) sensors. They'll have CAN, I2C, and UART available. Not wanting to re-invent the wheel, I'm wondering what the best approach would be from a protocol standpoint to integrate these sensors with PX4? For example, how could a user chose to use our INS as the primary source for IMU, GNSS, and attitude / position / velocity data? Or use our air data sensor for airspeed? I'm familiar with MAV Link and UAV CAN / Drone CAN, but I'm not sure if that would be the preferred approach from a bandwidth perspective or if the hooks are already available in PX4 to configure and use those data sources.

Thanks!
Brian

dagar, 2021-11-30:
Hi Brian,

Right now I'd definitely recommend UAVCAN/DroneCAN. Most of the raw sensor data cases are already handled. https://github.com/PX4/PX4-Autopilot/tree/master/src/drivers/uavcan/sensors

Properly integrating offboard estimates requires a bit more effort, but it's something I'm interested in for various reasons.

If you'd like to discuss in more detail you could join our weekly dev call (https://docs.px4.io/master/en/contribute/dev_call.html) or PX4 slack (https://join.slack.com/t/px4/shared_invite/zt-si4xo5qs-R4baYFmMjlrT4rQK5yUnaA).

AlexKlimaj, 2021-12-01:
CAN is definitely the easiest method as I2C or UART would require standalone drivers in PX4 that would need to be maintained. 
https://arkelectron.com/product-category/sensors/
The ultimate goal of my sensor products is to run the EKF and pass along the estimate over CAN via a uorb bridge.

flybrianfly, 2021-12-01:
Thanks, seems like we'll also use CAN

hamishwillee, 2025-10-29:
Closing as solved

closed 2025-10-29 by hamishwillee