PX4/PX4-Autopilot #18753: Pixhawk 4 - GPS Problem

FeCLN, 2021-11-30:
Hi, I am from Equipe Skyrats, a Brazilian autonomous drone team. We are having a problem with our GPS (M8N) when using it with a Pixhawk4. Sometimes it says ‘No GPS fix’, sometimes it says there’s ‘zero satellites’ when we know there are enough, and sometimes it says that the GPS signal is too weak. 
(Obs1: In the same time and location, these problems only occur with our Pixhawks; but with our Pixracers, using the same module, the GPS works fine, and with a good amount of signal ).
Has anyone had this same problem?
(Obs2: we already tested other Pixhawks and GPSs, and the same thing happens every time. The firmware version we use is always the PX4 Stable release. The tested GPSs in the Pixhawks were the Radiolink SE100 and the GPS that comes with the Pixhawk itself. And all this instability problems already occurred with us using other Pixhawks versions.)

dagar, 2021-11-30:
Can you share specific error messages or screenshots? Can you share a picture of the actual GPS module?

On the pixhawk can you open a console and run `gps status`?

Do you have anything else connected to this pixhawk? In particular things like USB 3 or certain radios can interfere with GPS.

hamishwillee, 2025-10-29:
Closing - more than just a little stale. If this is still a problem, please replicate on current main or stable and repost.