PX4/PX4-Autopilot #18759: Altitude keeps increasing

dookei, 2021-12-01:
Hi,
I noticed that when I leave the drone for a fews hours the altitude keeps increasing slowly.Even this is not flew and just sits on the table. If I leave it for a day running it evens reaches 100 meter.

Temperature seems to be within normal range and the barometer pressure is relatively constant between 990 ad 1005. 
So where does this continuous incremental is coming from ?

**NOTE**: if I restart the FC this goes back to 0. 

**Setup**
FC: Holybro Durandal
PX4: v1.12.3

```
nsh> listener sensor_baro

TOPIC: sensor_baro
 sensor_baro_s
    timestamp: 56678188008  (0.005302 seconds ago)
    timestamp_sample: 56678187997  (11 us before timestamp)
    device_id: 3997730 (Type: 0x3D, SPI:4 (0x00)) 
    error_count: 0
    pressure: 988.8800
    temperature: 35.2600
```

```
TOPIC: vehicle_imu_status 2 instances

Instance 0:
 vehicle_imu_status_s
    timestamp: 56420435519  (0.080898 seconds ago)
    accel_device_id: 3932170 (Type: 0x3C, SPI:1 (0x00)) 
    gyro_device_id: 3932170 (Type: 0x3C, SPI:1 (0x00)) 
    accel_clipping: [0, 0, 0]
    accel_error_count: 0
    gyro_error_count: 0
    accel_rate_hz: 399.6270
    gyro_rate_hz: 399.6270
    accel_raw_rate_hz: 3996.2700
    gyro_raw_rate_hz: 7992.5400
    accel_vibration_metric: 0.0002
    gyro_vibration_metric: 0.0000
    gyro_coning_vibration: 0.0000
    mean_accel: [-0.0707, -0.1210, -9.8343]
    mean_gyro: [-0.0009, 0.0006, 0.0000]
    temperature_accel: 38.5128
    temperature_gyro: 38.5128

Instance 1:
 vehicle_imu_status_s
    timestamp: 56420579318  (0.026744 seconds ago)
    accel_device_id: 6946826 (Type: 0x6A, SPI:1 (0x00)) 
    gyro_device_id: 6684682 (Type: 0x66, SPI:1 (0x00)) 
    accel_clipping: [0, 0, 0]
    accel_error_count: 0
    gyro_error_count: 0
    accel_rate_hz: 402.7789
    gyro_rate_hz: 399.3610
    accel_raw_rate_hz: 1611.1157
    gyro_raw_rate_hz: 1996.8052
    accel_vibration_metric: 0.0013
    gyro_vibration_metric: 0.0000
    gyro_coning_vibration: 0.0000
    mean_accel: [-0.1543, -0.1342, -9.7918]
    mean_gyro: [-0.0007, 0.0001, 0.0030]
    temperature_accel: 39.2500
    temperature_gyro: nan`
```

And here the parameters that I changed
```
nsh> param show -c
Symbols: x = used, + = saved, * = unsaved
x + BAT1_A_PER_V [1,24] : 36.3675
x + BAT1_N_CELLS [4,27] : 2
x + BAT1_SOURCE [6,29] : -1
x + BAT1_V_DIV [9,32] : 18.1000
x + BAT2_A_PER_V [12,35] : 36.3680
x + BAT2_N_CELLS [15,38] : 6
x + BAT2_SOURCE [17,40] : 0
x + BAT2_V_DIV [20,43] : 18.1000
x + BAT2_V_LOAD_DROP [22,45] : 4.0000
x + BAT_A_PER_V [25,49] : 36.3675
x + BAT_N_CELLS [30,55] : 2
x + BAT_SOURCE [32,58] : -1
x + BAT_V_DIV [34,60] : 18.1000
x + CAL_ACC0_ID [38,64] : 3932170
x + CAL_ACC0_PRIO [39,65] : 50
x + CAL_ACC0_XOFF [41,67] : -0.1522
x + CAL_ACC0_YOFF [43,69] : 0.1270
x + CAL_ACC0_ZOFF [45,71] : -0.1481
x + CAL_ACC1_ID [47,73] : 6946826
x + CAL_ACC1_PRIO [48,74] : 50
x + CAL_ACC1_XOFF [50,76] : -0.0323
x + CAL_ACC1_XSCALE [51,77] : 0.9929
x + CAL_ACC1_YOFF [52,78] : 0.0868
x + CAL_ACC1_YSCALE [53,79] : 0.9961
x + CAL_ACC1_ZOFF [54,80] : 0.1038
x + CAL_ACC1_ZSCALE [55,81] : 0.9927
x + CAL_GYRO0_ID [61,103] : 3932170
x + CAL_GYRO0_PRIO [62,104] : 50
x + CAL_GYRO0_XOFF [64,106] : 0.0163
x + CAL_GYRO0_YOFF [65,107] : 0.0134
x + CAL_GYRO0_ZOFF [66,108] : 0.0025
x + CAL_GYRO1_ID [67,109] : 6684682
x + CAL_GYRO1_PRIO [68,110] : 50
x + CAL_MAG0_ID [75,127] : 73225
x + CAL_MAG0_PRIO [76,128] : 75
x + CAL_MAG0_ROT [77,129] : 0
x + CAL_MAG0_XODIAG [79,131] : -0.0312
x + CAL_MAG0_XOFF [80,132] : -0.0610
x + CAL_MAG0_XSCALE [81,133] : 1.0759
x + CAL_MAG0_YODIAG [83,135] : -0.0454
x + CAL_MAG0_YOFF [84,136] : 0.1206
x + CAL_MAG0_YSCALE [85,137] : 1.0312
x + CAL_MAG0_ZODIAG [87,139] : -0.0188
x + CAL_MAG0_ZOFF [88,140] : -0.0185
x + CAL_MAG0_ZSCALE [89,141] : 0.8747
x + CAL_MAG1_ID [90,142] : 396825
x + CAL_MAG1_PRIO [91,143] : 50
x + CAL_MAG1_XODIAG [94,146] : -0.0294
x + CAL_MAG1_XOFF [95,147] : -0.3460
x + CAL_MAG1_XSCALE [96,148] : 1.0233
x + CAL_MAG1_YODIAG [98,150] : -0.0907
x + CAL_MAG1_YOFF [99,151] : -0.1636
x + CAL_MAG1_YSCALE [100,152] : 0.9325
x + CAL_MAG1_ZODIAG [102,154] : 0.0367
x + CAL_MAG1_ZOFF [103,155] : -0.3631
x + CAL_MAG1_ZSCALE [104,156] : 1.0606
x + COM_FLIGHT_UUID [147,229] : 11
x + COM_FLTMODE1 [148,230] : 8
x + COM_FLTMODE4 [151,233] : 8
x + COM_FLTMODE6 [153,235] : 2
x + COM_FLT_PROFILE [154,236] : 300
x + DSHOT_CONFIG [194,276] : 300
x + EKF2_AID_MASK [204,286] : 24
x + EKF2_IMU_POS_X [249,331] : -0.2100
x + LND_FLIGHT_T_LO [366,560] : 155211185
x + MAV_1_CONFIG [372,612] : 102
x + MAV_2_CONFIG [377,617] : 103
x + MAV_2_MODE [379,619] : 2
x + MAV_ODOM_LP [386,626] : 1
x + MC_PITCHRATE_D [403,643] : 0.0032
x + MC_PITCHRATE_I [405,645] : 0.1500
x + MC_PITCHRATE_K [406,646] : 0.5500
x + MC_PITCH_P [409,649] : 5.0000
x + MC_ROLLRATE_D [411,651] : 0.0036
x + MC_ROLLRATE_I [413,653] : 0.1500
x + MC_ROLLRATE_K [414,654] : 0.6500
x + MC_ROLLRATE_P [416,656] : 0.1660
x + MC_ROLL_P [417,657] : 4.5000
x + MC_YAWRATE_I [421,661] : 0.0800
x + MC_YAWRATE_K [422,662] : 1.1500
x + MC_YAWRATE_P [424,664] : 0.2400
x + MC_YAW_P [425,665] : 3.0000
x + NAV_RCL_ACT [507,767] : 0
x + PWM_AUX_OUT [517,816] : 123456
x + PWM_MAIN_OUT [553,947] : 12345678
x + RC1_MAX [619,1031] : 1994.0000
x + RC1_MIN [620,1032] : 994.0000
x + RC1_TRIM [622,1034] : 994.0000
x + RC2_MAX [624,1036] : 1994.0000
x + RC2_MIN [625,1037] : 994.0000
x + RC2_TRIM [627,1039] : 1492.0000
x + RC3_MAX [629,1041] : 2006.0000
x + RC3_MIN [630,1042] : 984.0000
x + RC3_TRIM [632,1044] : 1494.0000
x + RC4_MAX [634,1046] : 1994.0000
x + RC4_MIN [635,1047] : 994.0000
x + RC4_TRIM [637,1049] : 1492.0000
x + RC_CHAN_CNT [667,1079] : 18
x + RC_MAP_ARM_SW [674,1086] : 8
x + RC_MAP_FLTMODE [683,1095] : 6
x + RC_MAP_KILL_SW [685,1097] : 10
x + RC_MAP_LOITER_SW [686,1098] : 11
x + RC_MAP_OFFB_SW [689,1101] : 7
x + RC_MAP_PITCH [693,1105] : 3
x + RC_MAP_ROLL [697,1109] : 2
x + RC_MAP_THROTTLE [699,1111] : 1
x + RC_MAP_YAW [701,1113] : 4
x + SDLOG_PROFILE [723,1144] : 255
x + SENS_BOARD_ROT [728,1149] : 18
x + SENS_BOARD_X_OFF [729,1150] : 0.1046
x + SENS_BOARD_Y_OFF [730,1151] : 2.0147
x + SER_TEL2_BAUD [770,1220] : 460800
x + SER_TEL3_BAUD [771,1221] : 460800
x + SYS_AUTOSTART [773,1252] : 24002
x + SYS_RESTART_TYPE [788,1267] : 0
x + THR_MDL_FAC [798,1445] : 0.3000

 806/1501 parameters used.
```

[image]

bresch, 2021-12-02:
1005 to 990 mbar makes a bit more than 100m of height increase, so it doesn't look so strange to me.

dookei, 2021-12-02:
thanks for your reply @bresch 
Sorry my bad. I kept an eye for the last hours and the baro stays stable around 988 to 994. But I will leave it running for the day to see what happens. I also had disabled TC_B_ENABLE.
I have now enabled it. To be continued :-) 

This is the current state:
```
nsh> param show TC_B_ENABLE
Symbols: x = used, + = saved, * = unsaved
x + TC_B_ENABLE [828,1376] : 1

nsh> listener sensor_baro

TOPIC: sensor_baro
 sensor_baro_s
	timestamp: 97011421  (0.016192 seconds ago)
	timestamp_sample: 97011410  (11 us before timestamp)
	device_id: 3997730 (Type: 0x3D, SPI:4 (0x00)) 
	error_count: 0
	pressure: 994.6800
	temperature: 35.8300

nsh> listener vehicle_odometry

TOPIC: vehicle_odometry
 vehicle_odometry_s
	timestamp: 114052163  (0.005057 seconds ago)
	timestamp_sample: 114051458  (705 us before timestamp)
	x: -0.0060
	y: 0.0020
	z: -0.1937
```

dookei, 2021-12-02:
@bresch so i left it now running for the day

```
nsh> listener sensor_baro

TOPIC: sensor_baro
 sensor_baro_s
	timestamp: 13186450659  (0.008670 seconds ago)
	timestamp_sample: 13186450649  (10 us before timestamp)
	device_id: 3997730 (Type: 0x3D, SPI:4 (0x00)) 
	error_count: 0
	pressure: 997.4700
	temperature: 36.3900

nsh> listener vehicle_odometry

TOPIC: vehicle_odometry
 vehicle_odometry_s
	timestamp: 13094688431  (0.007727 seconds ago)
	timestamp_sample: 13094687756  (675 us before timestamp)
	x: -0.1537
	y: -0.1527
	z: 20.4687
	q: [-0.3787, -0.0001, 0.0024, 0.9255]  (Roll: 0.2 deg, Pitch: -0.1 deg, Yaw: -135.5 deg)
	q_offset: [0.0000, 0.0131, 0.9999, -0.0000]  (Roll: -180.0 deg, Pitch: 0.0 deg, Yaw: 178.5 deg)
	pose_covariance: [0.0030, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0030, 0.0000, 0.0000, 0.0000, 0.0000, 0.2848, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000]
	vx: -0.0048
	vy: -0.0051
	vz: -0.0067
	rollspeed: 0.0001
	pitchspeed: 0.0008
	yawspeed: -0.0014
	velocity_covariance: [0.0039, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0039, 0.0000, 0.0000, 0.0000, 0.0000, 0.0215, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000]
	local_frame: 0
	velocity_frame: 1
```
I will reboot it and start the log as you recommended. I will post the results tomorrow morning.

dookei, 2021-12-03:
I left it now running for the morning. For some reason the altitude is not incrementing now. 
I have a 300mb log(still downloading, the logger was not accepting lower rates just the default)which I assume there wont have  nothing abnormal. 

Before I was performing some tests with VIO. Perhaps is something to do with that?
```
TOPIC: sensor_baro
 sensor_baro_s
	timestamp: 12040601936  (0.017538 seconds ago)
	timestamp_sample: 12040601926  (10 us before timestamp)
	device_id: 3997730 (Type: 0x3D, SPI:4 (0x00)) 
	error_count: 0
	pressure: 1004.4200
	temperature: 37.1800
```

```
TOPIC: vehicle_odometry
 vehicle_odometry_s
	timestamp: 11938125423  (0.001715 seconds ago)
	timestamp_sample: 11938124685  (738 us before timestamp)
	x: 0.0015
	y: -0.0032
	z: -5.4275
```
Anyway, I will come back to you once I found some more details.

bresch, 2021-12-06:
> Before I was performing some tests with VIO. Perhaps is something to do with that?

Yes, that's quite likely

dookei, 2021-12-06:
Ok.  Thank you very much for your feedback.

closed 2021-12-06 by dookei