PX4/PX4-Autopilot #18760: Odometry message not displayed with VIO

dookei, 2021-12-01:
Hi there,
I am currently trying to setup a VIO system that is publish messages from mavros to Px4 over serial Port or USB.
This was tried in two different versions from PX4 (v1.11.3 and v.1.12.3) and in none of them seems to work.
I have set  MAV_ODOM_LP to 1 and EKF2_AID_MASK [204,286] : 24.
Ofcourse if I try to change to HOLD or POSITION mode this is not accepted. 
Any idea why this is happening ?

Under uorb I can see that the message is being publish properly.

**Ouput from vehicle_visual_odometry**
```
nsh> listener vehicle_visual_odometry

TOPIC: vehicle_visual_odometry
 vehicle_odometry_s
    timestamp: 57550407383  (0.004223 seconds ago)
    timestamp_sample: 57550375293  (32090 us before timestamp)
    x: 0.0057
    y: -0.0055
    z: -0.0278
    q: [0.3169, 0.0088, 0.0008, 0.9483]  (Roll: 0.4 deg, Pitch: -0.9 deg, Yaw: 143.0 deg)
    q_offset: [0.0000, 0.0000, 0.0000, 0.0000]  (Roll: 0.0 deg, Pitch: -0.0 deg, Yaw: 0.0 deg)
    pose_covariance: [0.0000, 0.0000, -0.0000, 0.0000, 0.0000, -0.0000, 0.0000, -0.0000, 0.0000, 0.0000, -0.0000, 0.0000, -0.0000, -0.0000, 0.0000, 0.0000, 0.0000, -0.0000, 0.0000, -0.0000, 0.0000]
    vx: nan
    vy: nan
    vz: nan
    rollspeed: nan
    pitchspeed: nan
    yawspeed: nan
    velocity_covariance: [nan, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000]
    local_frame: 0
    velocity_frame: 1
```
**Ouput from vehicle_odometry**, is taking also the coordinates from the VIO
```
TOPIC: vehicle_odometry
 vehicle_odometry_s
    timestamp: 57621717790  (0.003962 seconds ago)
    timestamp_sample: 57621716944  (846 us before timestamp)
    x: 0.0050
    y: -0.0075
    z: -96.5211
    q: [0.3169, 0.0057, 0.0043, 0.9484]  (Roll: 0.6 deg, Pitch: -0.4 deg, Yaw: 143.0 deg)
    q_offset: [0.0000, -0.9953, 0.0962, -0.0000]  (Roll: -180.0 deg, Pitch: -0.0 deg, Yaw: -11.0 deg)
    pose_covariance: [0.0008, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0008, 0.0000, 0.0000, 0.0000, 0.0000, 0.2846, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000]
    vx: -0.0039
    vy: -0.0025
    vz: 0.0311
    rollspeed: -0.0004
    pitchspeed: -0.0012
    yawspeed: 0.0012
    velocity_covariance: [0.0022, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0022, 0.0000, 0.0000, 0.0000, 0.0000, 0.0215, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000]
    local_frame: 0
    velocity_frame: 1
```

**NOTE**: I noticed that there are in some topics they are doubled and tripled. Is that normal ?
```
nsh> uorb status
TOPIC NAME                      INST #SUB #Q SIZE PATH
actuator_armed                     0   16  1   16 /obj/actuator_armed0
actuator_controls_0                0    9  1   48 /obj/actuator_controls_00
actuator_controls_3                0    3  1   48 /obj/actuator_controls_30
actuator_outputs                   0    5  1   80 /obj/actuator_outputs0
actuator_outputs                   1    2  1   80 /obj/actuator_outputs1
adc_report                         0    1  1   96 /obj/adc_report0
battery_status                     0   11  1  160 /obj/battery_status0
commander_state                    0    1  1   16 /obj/commander_state0
cpuload                            0    6  1   16 /obj/cpuload0
ekf2_timestamps                    0    1  1   24 /obj/ekf2_timestamps0
[... 119 lines cut ...]
vehicle_attitude_setpoint          0    6  1   64 /obj/vehicle_attitude_setpoint0
vehicle_command                    0   18  8   56 /obj/vehicle_command0
vehicle_command_ack                0    5  4   24 /obj/vehicle_command_ack0
vehicle_control_mode               0   21  1   24 /obj/vehicle_control_mode0
vehicle_gps_position               0    6  1  104 /obj/vehicle_gps_position0
vehicle_imu                        0    8  1   56 /obj/vehicle_imu0
vehicle_imu                        1    8  1   56 /obj/vehicle_imu1
vehicle_imu_status                 0    7  1   96 /obj/vehicle_imu_status0
vehicle_imu_status                 1    6  1   96 /obj/vehicle_imu_status1
vehicle_land_detected              0   16  1   24 /obj/vehicle_land_detected0
vehicle_local_position             0   23  1  168 /obj/vehicle_local_position0
vehicle_magnetometer               0    6  1   40 /obj/vehicle_magnetometer0
vehicle_magnetometer               1    3  1   40 /obj/vehicle_magnetometer1
vehicle_odometry                   0    0  1  256 /obj/vehicle_odometry0
vehicle_rates_setpoint             0    6  1   32 /obj/vehicle_rates_setpoint0
vehicle_status                     0   40  1   80 /obj/vehicle_status0
vehicle_status_flags               0    6  1   48 /obj/vehicle_status_flags0
vehicle_visual_odometry            0    8  1  256 /obj/vehicle_visual_odometry0
yaw_estimator_status               0    1  1  104 /obj/yaw_estimator_status0
yaw_estimator_status               1    1  1  104 /obj/yaw_estimator_status1
yaw_estimator_status               2    1  1  104 /obj/yaw_estimator_status2
yaw_estimator_status               3    1  1  104 /obj/yaw_estimator_status3
```
The message LOCAL_POSITION_NED is taking the VIO input.
[image]

dookei, 2021-12-06:
As far as I understand the EKF2 is accepting the vehicle_visual_odometry. This can be observed through the console/terminal.
The message ODOMETRY is only being streamed on the mavlink instance #0.
listener vehicle_odometry
listener vehicle_visual_odometry.

`nsh>   mavlink status streams

instance #0:
	Name                Rate Config (current) [Hz] Message Size (if active) [B]
	HEARTBEAT                       1.00 (1.000)    21
	STATUSTEXT                     20.00 (20.000)   66
	COMMAND_LONG                  unlimited       
	TIMESYNC                       10.00 (10.000)   28
	CAMERA_TRIGGER                unlimited       
	LOCAL_POSITION_NED             30.00 (30.000)   40
	DISTANCE_SENSOR                10.00 (10.000) 
	MOUNT_ORIENTATION              10.00 (10.000) 
	ODOMETRY                       30.00 (30.000)  244
	ACTUATOR_CONTROL_TARGET0       30.00 (30.000)   53
	ADSB_VEHICLE                  unlimited       
	ALTITUDE                       10.00 (10.000)   44
	ATTITUDE                       50.00 (50.000)   40
	ATTITUDE_QUATERNION            50.00 (50.000)   60
	ATTITUDE_TARGET                 8.00 (8.000)    49
	BATTERY_STATUS                  0.50 (0.500)    66
	CAMERA_IMAGE_CAPTURED         unlimited       
	COLLISION                     unlimited       
	ESC_INFO                       10.00 (10.000) 
	ESC_STATUS                     10.00 (10.000) 
	ESTIMATOR_STATUS                5.00 (5.000)    54
	EXTENDED_SYS_STATE              2.00 (2.000)    14
	GLOBAL_POSITION_INT            10.00 (10.000) 
	GPS2_RAW                      unlimited       
	GPS_GLOBAL_ORIGIN               1.00 (1.000)    32
	GPS_RAW_INT                   unlimited         64
	GPS_STATUS                      1.00 (1.000)  
	HIGHRES_IMU                    50.00 (50.000)   75
	HOME_POSITION                   0.50 (0.500)    72
	MANUAL_CONTROL                  5.00 (5.000)    23
	NAV_CONTROLLER_OUTPUT          10.00 (10.000) 
	OPTICAL_FLOW_RAD               10.00 (10.000) 
	ORBIT_EXECUTION_STATUS          5.00 (5.000)  
	PING                            1.00 (1.000)    26
	POSITION_TARGET_GLOBAL_INT     10.00 (10.000)   63
	RAW_RPM                         5.00 (5.000)  
	RC_CHANNELS                    10.00 (10.000)   54
	SCALED_IMU                     25.00 (25.000)   36
	SCALED_IMU2                    25.00 (25.000)   36
	SCALED_IMU3                    25.00 (25.000) 
	SERVO_OUTPUT_RAW_0             20.00 (20.000)   49
	SERVO_OUTPUT_RAW_1             20.00 (20.000)   49
	SYS_STATUS                      1.00 (1.000)    43
	SYSTEM_TIME                     1.00 (1.000)    24
	UTM_GLOBAL_POSITION             1.00 (1.000)  
	VFR_HUD                        20.00 (20.000)   32
	VIBRATION                       2.50 (2.500)    44
	WIND_COV                       10.00 (10.000) 
	DEBUG                          50.00 (50.000) 
	DEBUG_FLOAT_ARRAY              50.00 (50.000) 
	DEBUG_VECT                     50.00 (50.000) 
	NAMED_VALUE_FLOAT              50.00 (50.000) 
	LINK_NODE_STATUS                1.00 (1.000)    48
	AUTOPILOT_VERSION                inf (inf)      90
`

closed 2021-12-06 by dookei

Ali-barari, 2023-07-24:
I have the similar setup and facing same issue. What was the solution? When I switch to POS HOLD, it give "No valid local position" error.