PX4/PX4-Autopilot #18766: PX4 1.12.3 Autoland Sets Yaw setpoint to +/- 90 degrees instead of current heading

gssammy, 2021-12-01:
**Describe the bug**
A clear and concise description of the bug.
Setting mode to Autoland from Offboard causes the drone to yaw to 180 and then yaw back to the current setting while landing. This happens both in sitl simulation and real flying. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start sending waypoints to drone
2. Arm drone and changed mode to Offboard
3. While drone is following commanded waypoints switch mode to AutoLand
4. See drone do weird yaws while descending

**Expected behavior**
The drone should go straight down and land

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
Ulog from real flight:
https://logs.px4.io/plot_app?log=90c95b8f-47be-41b4-a508-4609fe3b3edf#Nav-Yaw-Angle

Ulog from sitl with custom gazebo setup:
https://logs.px4.io/plot_app?log=5bf30ece-9637-4a9a-b847-230927c8159f#Nav-Yaw-Angle

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
This is a PX4 quadcopter
Real drone uses a pixracer

**Additional context**
- This did not happen in 1.11.3 with the same setup.
- This does not happen in offboard loss (triggered by killing the waypoint publisher)

It seems that there is something in `land.cpp` that is setting the yaw of the waypoint before landing to 180. I edited `land.cpp` with some print outs just to see what was happening:

```
Land::on_activation()
{
	/* set current mission item to Land */
	set_land_item(&_mission_item, true);
	_navigator->get_mission_result()->finished = false;
	_navigator->set_mission_result_updated();
	reset_mission_item_reached();

	/* convert mission item to current setpoint */
	struct position_setpoint_triplet_s *pos_sp_triplet = _navigator->get_position_setpoint_triplet();
	printf("Current Yaw: %9.6f" ,pos_sp_triplet->current.yaw);
	pos_sp_triplet->previous.valid = false;
	mission_apply_limitation(_mission_item);
	mission_item_to_position_setpoint(_mission_item, &pos_sp_triplet->current);
	pos_sp_triplet->next.valid = false;

	_navigator->set_can_loiter_at_sp(false);

	_navigator->set_position_setpoint_triplet_updated();
	printf("New Yaw: %9.6f",pos_sp_triplet->current.yaw);
	mavlink_log_info(&_mavlink_log_pub, "i changed waypoints");

}
```
and in the sitl terminal this is what i see:
```
INFO  [commander] Armed by external command
INFO  [commander] Takeoff detected
Current Yaw:  0.000000New Yaw:  1.570682INFO  [navigator] i changed waypoints
[ WARN] [1638329903.252801509, 47.616000000]: CMD: Unexpected command 176, result 0
INFO  [commander] Landing at current position
```

gssammy, 2021-12-01:
using the stock px4 sitl with gazebo (`roslaunch px4 mavros_posix_sitl.launch`) and sending the waypoints
```
rostopic pub -r 10 /mavros/setpoint_position/local geometry_msgs/PoseStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
pose:
  position:
    x: 0.0
    y: 0.0
    z: 3.0
  orientation:
    x: 0.0
    y: 0.0
    z: 1.0
    w: 0.0" 
```
i see the effect but not as bad: https://logs.px4.io/plot_app?log=52625e87-0565-4381-b2ee-bead9eb4c2d5#Nav-Yaw-Angle.

Full terminal output:
```roslaunch px4 mavros_posix_sitl.launch 
... logging to /home/george/.ros/log/011dc2b6-52b9-11ec-80fc-5b7f4fcc37e0/roslaunch-george-pensa-23686.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://george-pensa:36293/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
[... 360 lines cut ...]
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative
[ INFO] [1638371527.380049541, 16.544000000]: RP: mission received
[ INFO] [1638371527.380477882, 16.544000000]: WP: mission received
[ INFO] [1638371527.380737406, 16.544000000]: GF: mission received
pxh> commander mode offboard
pxh> [ WARN] [1638371533.052255230, 22.208000000]: CMD: Unexpected command 176, result 0
pxh> commander arm
pxh> INFO  [commander] Armed by internal command
INFO  [commander] Takeoff detected
pxh> commander mode auto:land
pxh> Current Yaw:  0.000000New Yaw: -1.570360INFO  [navigator] i changed waypoints
[ WARN] [1638371554.463229513, 43.484000000]: CMD: Unexpected command 176, result 0
INFO  [commander] Landing detected
INFO  [commander] Disarmed by landing
INFO  [logger] closed logfile, bytes written: 3438881
pxh> shutdown
```

AlexisTM, 2024-10-22:
You are sending a setpoint with an orientation. Furthermore, this is not the unary quaternion (which should be w = 1, not z = 1)

closed 2024-10-23 by bresch