PX4/PX4-Autopilot #18774: MacOS Build v1.12.3 issue: module future not found

jatast, 2021-12-03:
Describe the bug
Unable to build latest firmware on Mac. Am thrown the "module future not found" Output is below. Have followed toolchain installation steps for px4 according to docs and have been able to build previous px4 versions (v12.0) perfectly fine on Mac. I suspect that this has to do with the new way of generating the mavlink libraries inside the mavlink module. When trying to generate mavlink library separately on mac i get the same error. I see there is no support for mavlink generation on Mac also. Only linux and windows.

[0/1] Re-running CMake...
-- PX4 version: v1.13.0-alpha1-3658-g2190f66096
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.6", minimum required is "3") 
-- PX4 config file: /Users/jacquesstoop/repos/PX4-Autopilot/boards/px4/fmu-v3/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m4
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- CONSTRAINED_MEMORY y
-- SERIAL_GPS1 /dev/ttyS3
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS6
-- UAVCAN_INTERFACES 2
-- PX4 config: px4_fmu-v3_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- drivers/px4io: ROMFS including px4_io-v2_default
CMake Deprecation Warning at src/drivers/uavcan/libuavcan/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/drivers/uavcan/libuavcan/libuavcan/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /usr/local/bin/python3 (found version "3.9.6") 
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/px4/fmu-v3/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/px4/fmu-v3/extras/px4_fmu-v3_bootloader.bin -> /etc/extras/px4_fmu-v3_bootloader.bin
-- ROMFS:  Adding boards/px4/fmu-v3/extras/px4_io-v2_default.bin -> /etc/extras/px4_io-v2_default.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jacquesstoop/repos/PX4-Autopilot/build/px4_fmu-v3_default
[0/1068] Generating Mavlink developmen...ssage_definitions/v1.0/development.xml
Traceback (most recent call last):
  File "/Users/jacquesstoop/repos/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/Users/jacquesstoop/repos/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ModuleNotFoundError: No module named 'future'
[0/1068] Generating ../../../platforms/nuttx/NuttX/nuttx/.config
FAILED: mavlink/development/development.h /Users/jacquesstoop/repos/PX4-Autopilot/build/px4_fmu-v3_default/mavlink/development/development.h 
cd /Users/jacquesstoop/repos/PX4-Autopilot/build/px4_fmu-v3_default/src/modules/mavlink && /usr/local/bin/python3 /Users/jacquesstoop/repos/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py --lang C --wire-protocol 2.0 --output /Users/jacquesstoop/repos/PX4-Autopilot/build/px4_fmu-v3_default/mavlink /Users/jacquesstoop/repos/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
[2/1068] Generating ../../../platforms/nuttx/NuttX/nuttx/.config
ninja: build stopped: subcommand failed.
make: *** [px4_fmu-v3] Error 1

To Reproduce
Install MacOS toolchain according to documentation
Clone latest repo
make any px4 target

Additional context
Running on latest MacOS M1 chip using Rosetta terminal

julianoes, 2021-12-03:
> ModuleNotFoundError: No module named 'future'

`pip3 install --user future`

jatast, 2021-12-03:
Ah my mistake found the problem was these lines in my .zshrc file pointing pip to the wrong python folder

# Point pip3 to MacOS system python 3 pip
alias pip3=/usr/bin/pip3

closed 2021-12-03 by jatast