PX4/PX4-Autopilot #18775: Changing a PWM channel

Ador2, 2021-12-03:
I'm using PX4 with fixed wing airframe, I want to change the throttle channel for example to be channel 6.
is that possible using parameters? or what should i do?

dagar, 2021-12-03:
Currently it's done by setting the mixer in an airframe. 

https://github.com/PX4/PX4-Autopilot/blob/4e3d944f58f0d60e072641fc60b426fbcfdad8ab/ROMFS/px4fmu_common/init.d/airframes/2100_standard_plane#L29

https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/mixers/AETRFG.main.mix

So for right now you can either change the `set MIXER AETRFG` or pick a different airframe that matches your desired output.

We're currently working on a complete overhaul of the output system that will allow you to directly modify everything via parameters or a GUI. https://github.com/mavlink/qgroundcontrol/pull/9952

Ador2, 2021-12-04:
I'm using AAVVTWFF.main.mix actually but for AETRFG.main.mix if I were to copy the content of CH3 [here](https://github.com/PX4/PX4-Autopilot/blob/4e3d944f58f0d60e072641fc60b426fbcfdad8ab/ROMFS/px4fmu_common/mixers/AETRFG.main.mix#L40) and overwrite CH6 with it, that would consider channel 3 and 6 for throttle, since my CH3 is dead I can use CH6 instead?
Also, to make sure, those are PWM channels right?

Ador2, 2021-12-04:
Yeah that works perfectly, thanks

closed 2021-12-04 by Ador2