PX4/PX4-Autopilot #18781: Yaw offset in autoland when flying with with external vision pose

marcelino-pensa, 2021-12-03:
**Describe the bug**
I am using px4 1.12.3 and I am having yaw (heading) issues in autoland when flying a quadcopter. There's always some yaw as soon as Autoland is triggered, followed by a yaw ramp change (as shown in the image below). We have observed this issue both in our physical drone, as well as in simulation (sitl):
[image]

I have spent the day yesterday trying to trace where this was coming from, and I think that I found out what the problem was. Below I describe the pipeline that leads to that:
1) As soon as we boot px4, it starts estimating its own heading/yaw. We have the magnetometer disabled (`EKF2_MAG_TYPE = 5`), so I'm not entirely sure how yaw is being estimated (probably some dead reckoning, which drifts over time).
2) Whenever we start sending vision pose data to the drone, it locks itself into the incoming yaw from the external vision data. The file `src/modules/ekf2/EKF2.cpp` records the offset correction between yaw before and after starting vision pose. Most of the time this offset is around 90deg for our flights, not sure why
3) We fly a mission, and then trigger autoland. As soon as we trigger autoland, the offset computed in part `2)` above is applied again whenever `FlightTask::update` is called when autoland is triggered. I have noticed that this function is called in a loop within `src/modules/flight_mode_manager/tasks/AutoMapper/FlightTastAutoMapper.cpp` when we go to autoland.

Now I'm wondering whether that's a desired use case, of whether it is a bug in the latest release. I could not find any ways to fix this issue other than changing the code and recompiling onto our system.

**To Reproduce**
- Start a quadcopter in release 1.12.3
- Rotate it until the estimated heading is nonzero
- Start vision_pose, sending position+heading data from an external vision source
- Hover (I've been doing this in offboard mode, not sure if the issue also happens out of offboard mode)
- Trigger autoland
- Drone will yaw while landing, then ramp down to the initial yaw from when it started autoland

**Expected behavior**
Descend in autoland without yaw

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- https://review.px4.io/plot_app?log=1deb2acc-b8b8-4132-9c7b-5e508657f0fd

**Drone (please complete the following information):**
- Describe the type of drone: quadcopter in X
- Photo of the IMU / autopilot setup if possible: this happens with our Pixracer, and in simulation as well (gazebo sitl)

**Additional context**
I suspect that the issue might be the same one observed in https://github.com/PX4/PX4-Autopilot/issues/18391. It is different because they did not observe it in autoland, but rather in normal flight. I think that it can be related because there is a counter there that only triggers the reset if the counter is at zero. I think that it is possible that the counter eventually overflows and returns to zero again, causing what was observed in this other issue (I'm not sure of that, it's just a personal hunch)

beniaminopozzan, 2022-09-20:
@marcelino-pensa , I'm having exactly the same issue (with release 1.13.0_dev), did you solve it in the end?

z1047941150, 2023-03-16:
> @marcelino-pensa , I'm having exactly the same issue (with release 1.13.0_dev), did you solve it in the end?

I'm having exactly the same issue (with release 1.13.2), did you solve it in the end?

beniaminopozzan, 2023-03-16:
@z1047941150 no, I didn't have time to investigate it further.