PX4/PX4-Autopilot #18782: How can I more thoroughly trace the build process of PX4?

edsut, 2021-12-03:
I've posted questions to the forum, figured I'd take a shot here...
I'm new to PX4, attempting a port to a very crude Teensy4.1 based flight controller hack I am working on.
I'm trying to understand how everything is built (for example builtin_list.[ch] and the rcS file).  Is
there any way to enable more trace at "make" time?  I've been using the following make command:
**export VERBOSE=1; make V=1 nxp_teensyhack_default 2>&1 | tee make.out**
but that doesn't do much more than show me all the compilation lines.
I also added **@$(eval CMAKE_ARGS += -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON)** and forced **PX4_MAKE_ARGS := -v**
in the Makefile but that didn't seem to help.
Any docs or suggestions?

dagar, 2021-12-03:
Typically what I do is go into the build directory and run things verbose single threaded. If you have the default tools installed ninja-build is used instead of make.

#### Example
``` Console

$ make px4_fmu-v5x_default
-- PX4 version: v1.13.0-alpha1-3664-g1f3391f658
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/dagar/git/PX4-Autopilot/boards/px4/fmu-v5x/default.px4board
...
...
...
[1070/1072] Linking CXX executable px4_fmu-v5x_default.elf
Memory region         Used Size  Region Size  %age Used
      FLASH_ITCM:          0 GB      2016 KB      0.00%
      FLASH_AXIM:     1961713 B      2016 KB     95.03%
        ITCM_RAM:          0 GB        16 KB      0.00%
        DTCM_RAM:          0 GB       128 KB      0.00%
           SRAM1:       90868 B       368 KB     24.11%
           SRAM2:          0 GB        16 KB      0.00%
[1072/1072] Creating /home/dagar/git/PX4-Autopilot/build/px4_fmu-v5x_default/px4_fmu-v5x_default.px4

$ cd build/px4_fmu-v5x_default/
$ ninja clean
[1/1] Cleaning all built files...
Cleaning... 1453 files.
$ ninja -v -j1
```

If I want to see more NuttX specific details then I go directly into the NuttX directory and fallow regular NuttX instructions (after PX4 configure).

``` Console

$ make px4_fmu-v5x_default
$ cd platforms/nuttx/NuttX/nuttx/
$ make V=1
```

edsut, 2021-12-10:
Thanks, sorry for late reply, I thought I would get an email notification.  
Anyway... unfortunately, after ninja clean, the builtin_list.h file is still there, so that doesn't help.
I've since learned (from the forum) that this file is at least partially created by platforms/nuttx/NuttX/CMakeLists.txt.  The builtins are apparently created based on the content of default.cmake in the target directory; but then other builtins seem to get added based on different settings in menuconfig (for example the builtin 'sercon' is included if I set CONFIG_SYSTEM_CDCACM=y).  
Problem is I need to change the builtin list, but when I modify the default.cmake file I then get compile errors. 
Obviously I have not yet gotten my head around this build process.  If there is *any* text that gives a little idea on the philosophy of the build that would really be helpful.

danduril, 2021-12-12:
`$ ninja -v -j1 2>&1 | tee mylog.txt` is an old standby, too.

dagar, 2021-12-12:
@edsut it can be a bit confusing because there's a NuttX build within the PX4 build. For the most part it's a regular standalone NuttX Makefile build (somewhat hidden behind cmake), but there are a few places we need to get in to extend the system. One example is the builtin_list you've stumbled across. We don't touch this file directly, but we insert the data NuttX uses to generate it (px4.bdat and px4.pdat).

hamishwillee, 2025-10-29:
Closing. Looks like it was addressed way back when.

closed 2025-10-29 by hamishwillee