PX4/PX4-Autopilot #18794: Switching to hold is currently not available. But Position works

dookei, 2021-12-06:
I am currently feeding with VIO over a USB port(/dev/ttyACM0)  to the PX4. I am able to fly in stabilized and position mode.
When in **Position Mode** and the sticks are in the middle position the drone is able to keep is position just by using the VIO.
Although, when I try to change to **HOLD mode** I get: **WARNING [commander] Switching to hold is currently not available.**

Since position is able to "hold" why is Hold mode not being accepted ? Log is attached below.

Setup: HW:Holybro Durandal SW:px4 v1.12.3 
Airframe: tri-clustered dodecarotor

**Expected behavior**
switch to HOLD.

**Log File**
https://logs.px4.io/plot_app?log=3e7cb6e7-5b1a-48a7-9292-005c89b7999e

bresch, 2021-12-07:
@dookei This is because "Hold" is an auto mode that requires a valid global position.
https://github.com/PX4/PX4-Autopilot/blob/d0d7f294223489bf7e586b7e124e5adaaf6f151e/src/modules/commander/state_machine_helper.cpp#L350-L357

This is because while the multicoper guidance is done in the local frame, the high-level mission interface and the fixedwing guidance are working in the global coordinates only. Therefore you cannot switch into an auto mode without a valid global position (except for takeoff that has a special case handling and only requires a valid local position).

dookei, 2021-12-09:
@bresch thanks once again for your feedback. What about if I also add ||  status_flags.condition_local_position_valid) ?
https://github.com/PX4/PX4-Autopilot/blob/020215f1462b1ed32d1dbc4f9c46ff11c03788d5/src/modules/commander/state_machine_helper.cpp#L302

What would be the consequences of that?  Since MAIN_STATE_POSCTL also uses it?

bresch, 2021-12-09:
> What about if I also add || status_flags.condition_local_position_valid) ?

You would need to check. I'm not sure about the consequences for fixedwings, rovers, ...

dookei, 2021-12-09:
Ok I will test on my multirotor. Rover and Fixedwing can only test in SIL,

closed 2021-12-09 by dookei

dookei, 2023-03-30:
Hi @mortenfyhn it has been a while since I tried that. At the time we changed our system to fully offboard mode control. Since we where using either way route planing, collision avoidance, precision landing etc on the Jetson system.
Sorry that I can't help you any further but I switched jobs in between and I have no way to test that at the moment.

mortenfyhn, 2023-03-30:
I understand, thank you anyway for responding!