PX4/PX4-Autopilot #18799: Plane airspeed sensor failure when taking off - crashed in manual mode

tuanpfiev, 2021-12-07:
**Describe the bug**
When the plane was taking off, the following warning was given:
_WARN [airspeed_selector] Airspeed sensor failure detected (1, -1)._
The pilot was using the manual mode at that moment. After about 1 minute, the airplane auto-disarmed while it was still in the air, and it crashed.  

We did several prior tests on the same day (with **a blank mission list**) and there was no issue. Then, we uploaded a mission to the airplane and this issue happened. The airplane was in Manual mode the whole time.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=f7e9a824-e8bb-419d-a084-36574ce042f5

A similar reported issue:
https://github.com/PX4/PX4-SITL_gazebo/issues/713

sfuhrer, 2021-12-07:
Hi @tuanpfiev 
Sorry to hear about your crash, hope the damage is reparable! 
We had these false-positive airspeed sensor failure detections quite often on fixed-wing vehicles, the logic isn't robust in case of manual starts. That was addressed in the mean-time.
As the system was assuming that the airspeed was invalid (wrongly), it relied GPS velocity (horizontal and vertical) and horizontal acceleration for the land detection. You got unlucky as there was a 2s phase in your flight with all 3 values below the thresholds:
[image]

I think it would make sense to tighten the thresholds in the airspeed-less case to prevent crashes as you had, see my PR https://github.com/PX4/PX4-Autopilot/pull/18806.

tuanpfiev, 2021-12-08:
@sfuhrer Thanks for your reply! Do you have any idea why that "unlucky" 2s phase happened? As I mentioned above, this issue only happened after I uploaded a mission to the autopilot (in this mission, the takeoff location was not defined. Could this be an issue?). 

In this flight test, the value of COM_DISARM_LAND = 2s. If I change this value to -1, the autopilot should not be disarmed even 
if the airspeed failure happens, right? Should I change something else so that the pilot can still control the aircraft in manual mode when something happens?

Below are some other log files that may be helpful for debugging.

**Manual takeoff->Stabilized mode->Loiter-> Landing (nominal)**
https://logs.px4.io/plot_app?log=004181bc-9035-4b3d-91dc-ba7f7a4aa8f2

**Manual takeoff (airspeed failure)**
https://logs.px4.io/plot_app?log=e1277fd0-8a79-4027-af9e-9c7b396ced7f

By the way, how did you make those plots?

sfuhrer, 2021-12-09:
> @sfuhrer Thanks for your reply! Do you have any idea why that "unlucky" 2s phase happened? As I mentioned above, this issue only happened after I uploaded a mission to the autopilot (in this mission, the takeoff location was not defined. Could this be an issue?).

No I think that was just bad coincidence. It is unlucky in the sense that you seemed to have quite a bit of wind, and some time after the airspeed "failure" when the vehicle was facing into the wind the groundspeed dropped below the threshold of 5m/s for more than 2s, while also not climbing nor experiencing high lateral accelerations. With the changes I proposed in https://github.com/PX4/PX4-Autopilot/pull/18806/files it wouldn't have happened. 

> In this flight test, the value of COM_DISARM_LAND = 2s. If I change this value to -1, the autopilot should not be disarmed even if the airspeed failure happens, right? Should I change something else so that the pilot can still control the aircraft in manual mode when something happens?

Yes correct, you can increase the threshold to some 10s or so or disable it, then the only ways to disarm is via stick gesture or "Disarm" command on the groundstation.

> By the way, how did you make those plots?

PlotJuggler, check it out, I can only recommend if you want to do in-depth log analysis: https://docs.px4.io/master/en/log/flight_log_analysis.html

closed 2021-12-10 by dagar