PX4/PX4-Autopilot #18822: kakute_f7 cannot use other flight mode than manual\stabilized\acro

911450448, 2021-12-09:
Hi,
I upload px4-firmware(1.13) on my kakute_f7,but i cannot used any flight mode without manu\stab\acro
I took holybro gps and qmc5883l on my flight controller and make sure they are set correctly,the problem remains.

No altitude data in QGC.

Did i do something wrong?

Thanks!
[image]

bresch, 2021-12-09:
Can you check that you have
- `SYS_HAS_BARO` = 1
- `SYS_HAS_MAG` = 1
- `SYS_MC_EST_GROUP` = 2

If you share a log file, we might be able to help you a bit more.

911450448, 2021-12-10:
> 你能检查一下你是否有
> 
> * `SYS_HAS_BARO` = 1
> * `SYS_HAS_MAG` = 1
> * `SYS_MC_EST_GROUP` = 2
> 
> 如果您共享日志文件，我们可能会为您提供更多帮助。

I checked setting,SYS_HAS_BARO = 1 and SYS_HAS_MAG = 1,but SYS_MC_EST_GROUP = Q attitude estimator.
And i checked altitude uORB,there are many N/A data,altitude data is not rought.

[image]

I will print my log file in a faw time.
thanks @bresch~

911450448, 2021-12-10:
> Can you check that you have
> 
> * `SYS_HAS_BARO` = 1
> * `SYS_HAS_MAG` = 1
> * `SYS_MC_EST_GROUP` = 2
> 
> If you share a log file, we might be able to help you a bit more.

Hi,i find my SYS_MC_EST_GROUP setting is 3 , in my default.cmake ekf2 is disable. So, if i changed this setting , should i enable ekf2 in my cmake? 
When i enable ekf2 ,kakute_f7 Flash_used is 112.6%.

I checked setting,SYS_HAS_BARO = 1 and SYS_HAS_MAG = 1.
And i checked altitude uORB,there are many N/A data,altitude data is not rought.

[image]

I will print my log file in a faw time.
thanks @bresch~

911450448, 2021-12-10:
News~
I modification file ~/kakutef7/default.cmake, enable ekf2 in my program, it seems that I can choose altitude mode , but when I arming my vehicle, I have a new problem...Way QGC report me:<Fail:No CPU load information>?  :(

bresch, 2021-12-10:
So the kakute F7 board doesn't have enough flash space to load the EKF2 and all the other modules and drivers. I assume that you had to remove something to make it fit?

dagar, 2021-12-10:
> So the kakute F7 board doesn't have enough flash space to load the EKF2 and all the other modules and drivers. I assume that you had to remove something to make it fit?

Maybe we should revisit https://github.com/PX4/PX4-ECL/pull/995?

911450448, 2021-12-11:
I enabled CONSTRAINED_FLASH setting like fmu-v2, and disable <battery status/… hot
/osd/top> and some other functions,it used 99.97% flash.
When I upload to kakute _f7,everything looks good, until I click ARM button, QGC report error.

a-altug, 2021-12-12:
Hi, @911450448 
I was stuck on that situation too.
Take a look:  [https://discuss.px4.io/t/bare-minimum-configuration-to-include-ekf2/16858/12]()

dagar, 2021-12-12:
If this is a common request we can find a way to either squeeze it into kakufef7 default (stripped down) or a special build.

911450448, 2021-12-13:
It`s working that I disable COM_CPU_MAX,everything right...I thing.
Now, I can arm my drone ,but I haven`t tested the other function yet,I hope everything good to use.

shupx, 2022-10-20:
> It`s working that I disable COM_CPU_MAX,everything right...I thing. Now, I can arm my drone ,but I haven`t tested the other function yet,I hope everything good to use.

Can you use OFFBORAD mode with kakute f7? Thanks! @911450448

AuroraRAS, 2023-02-03:
> No CPU load information

set the parameter `COM_CPU_MAX` to `-1` is working for me

hamishwillee, 2025-10-29:
Clossing - appears to have been solved.

closed 2025-10-29 by hamishwillee