PX4/PX4-Autopilot #18828: Not receving any data by /mavros/imu/data_raw

bhaskar-glitch, 2021-12-10:
Hi,
I'm using pixhawk 2.4.8 with px4 latest firmware now while launching the px4.launch file I'm not getting any data with topic **/mavros/imu/data_raw**(rostopic echo /mavros/imu/data_raw) although when I echo the **/mavros/imu/data** then it's showing the value of gyro only not the accelero and I've use the data of both gyro and accelro to sync and calibrate my camera with it..PLease help

Jaeyoung-Lim, 2022-02-22:
@bhaskar-glitch This simply means that you don't have connection to the fmu through mavros (mavros not receiving any mavlink messages)

bhaskar-glitch, 2022-02-23:
> @bhaskar-glitch This simply means that you don't have connection to the fmu through mavros (mavros not receiving any mavlink messages)

But whenever i use to stop my odometry data from publishing,the mavros/local_position/pose and odom also stops streaming.

Jaeyoung-Lim, 2022-02-23:
@bhaskar-glitch If you dont have a valid local position estimate, local position estimates it will stop streaming.