PX4/PX4-Autopilot #18829: unable to change mission flight mode

SamLin55587, 2021-12-10:
when I run multiple vehicle in SITL,they can arm but can't start mission.
But I run single vehicle simulation,it can start mission
I want to know how to fix this problem,Thanks.
[image]

Jaeyoung-Lim, 2021-12-10:
It appears that you are not using rhe default models/scripts to run multivehicle simulations

How are you running the multivehicle simulation simulation and how can we reproduce it?

SamLin55587, 2021-12-12:
But when I run the default model (standard plane)  this problem still happen
I don't know what's wrong with this

muhammadadeelzahid, 2022-08-19:
Hello. Is this resolved? can you share details on how to fix this