PX4/PX4-Autopilot #18855: [offboard] MPC_ACC_UP_MAX constraint is violated in offboard mode

squizz617, 2021-12-15:
When simulating `px4_sitl_rtps` firmware in Gazebo, trajectory_setpoint messages published via ROS2 rtps can make acceleration of the drone violate the `MPC_ACC_UP_MAX` constraint. Checking `param show -a`, parameter `MPC_ACC_UP_MAX` is being used, and the value is 4.0 (default). However, the vertical acceleration upwards gets larger than the parameter value (see the graphs below).

**To Reproduce**
Steps to reproduce the behavior:
1. [terminal 1] `make px4_sitl_rtps gazebo -j4`
2. [terminal 2] `micrortps_agent -t UDP`
3. [terminal 3] Publish the following trajectory_setpoint messages 50 times, once every 200 ms:
```
msg = TrajectorySetpoint()
msg.z = -10.0
msg.yaw = 0.0 
msg.acceleration[2] = -3000.0
```

**Expected behavior**
Huge acceleration setpoint should be clamped to the value set by `MPC_ACC_UP_MAX` parameter.

**Log Files and Screenshots**
* `z` of `vehicle_acceleration` uORB topic:
[image]
(-9.8 seems to be some sort of offset. Still, the maximum is -15.22, meaning that the acceleration gets as big as -5.42 m/s^2)
* `z`, `vz`, and `az` of `vehicle_local_postion` uORB topic:
[image]
Consistent result published to another topic. We can clearly see that `az` violates the constraint (i.e., |az| > 4.0).

In the controller code, I could not find any constraining logic using `MPC_ACC_UP_MAX` or `MPC_ACC_DOWN_MAX` for vertical acceleration. Are those parameters ever used?

bresch, 2021-12-15:
Hi @squizz617 ,

> z of vehicle_acceleration uORB topic [...] -9.8 seems to be some sort of offset.

`vehicle_acceleration` contains the low-passed accelerometer values (body frame). This is why there is an offset of 1G.

> In the controller code, I could not find any constraining logic using MPC_ACC_UP_MAX or MPC_ACC_DOWN_MAX for vertical acceleration. Are those parameters ever used?

Those parameters are used by the trajectory generators in PX4 (e.g.: https://github.com/PX4/PX4-Autopilot/blob/3fba7288af2fb99dbed755d77d7f5cf23b9ebf95/src/modules/flight_mode_manager/tasks/ManualPositionSmoothVel/FlightTaskManualPositionSmoothVel.cpp#L126-L135), not by the controllers. When using offboard, you need to generate your own trajectories satisfying the constraints you want, the controller just makes sure that its thrust output is between  [MPC_THR_MIN](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#MPC_THR_MIN) and [MPC_THR_MAX](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#MPC_THR_MAX).

squizz617, 2021-12-15:
Thanks @bresch for the clarification!
I have a follow-up question, then.

In the [parameter reference](https://docs.px4.io/master/en/advanced_config/parameter_reference.html), `MPC_ACC_UP_MAX` is described as "Maximum vertical acceleration in velocity controlled modes upward", and I thought offboard is one of the velocity controlled modes, from the following documentations and code:
* https://docs.px4.io/master/en/getting_started/flight_modes.html#offboard-mode-mc
* https://docs.px4.io/master/en/flight_modes/offboard.html#copter-vtol
* https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/commander/Commander.cpp#L3232-L3251

Also, as I mentioned in the original post, px4 reported that the parameters are in use:
```
pxh> param show -a
Symbols: x = used, + = saved, * = unsaved
...
x   MPC_ACC_DOWN_MAX [396,685] : 3.0000
...
x   MPC_ACC_UP_MAX [399,688] : 4.0000
...
```

But if those parameters are used only by the trajectory generators, aren't their descriptions in the parameter reference misleading?

Jaeyoung-Lim, 2021-12-15:
@squizz617 Offboard mode is an interface for prototyping your controllers for developers. Therefore it is technically not a "mode" since it is basically disabling all the safety/logic handling and enabling you to send setpoints directly to each of the cascaded control loops such as position, velocity, acceleration. 

If you are using acceleration setpoints, you are basically bypassing the velocity control loop. Also I noticed that you are sending a z position, yaw=0 and not the other values. This is NOT a combination of setpoints we support. You can check the combinations of type masks that PX4 supports: https://docs.px4.io/master/en/flight_modes/offboard.html#copter-vtol

squizz617, 2021-12-15:
Thanks @Jaeyoung-Lim for the reply!

> Offboard mode is an interface for prototyping your for developers. Therefore it is technically not a "mode" since it is basically disabling all the safety/logic handling and enabling you to send setpoints directly to each of the cascaded control loops such as position, velocity, acceleration.

Now I understand the intention of this mode. Still, I think the above should really be clarified in the documentation. Many places in the documentation, offboard mode is listed as one of the flight modes.
- as per [Getting started/flight_modes](https://docs.px4.io/master/en/getting_started/flight_modes.html#autonomous-and-manual-modes), offboard is an **autonomous mode**
- in [flying/flight_modes](https://docs.px4.io/master/en/flight_modes/) as well.

Also, the [offboard section](https://docs.px4.io/master/en/flight_modes/offboard.html#description) describes the mode as "Offboard mode is primarily used for controlling vehicle movement and attitude, and supports only a very limited set of MAVLink messages (more may be supported in future).", not mentioning anywhere that those parameters will be ignored.

> Also I noticed that you are sending a z position, yaw=0, and z acceleration and not setting or defining the other values. This is NOT a combination of setpoints we support.

Other values are set to defaults and omitted for brevity. The entire message is:
```
px4_msgs.msg.TrajectorySetpoint(
    timestamp=1021969620,
    x=0.0, y=0.0, z=-10.0,
    yaw=0.0, yawspeed=0.0,
    vx=0.0, vy=0.0, vz=0.0,
    acceleration=array([    0.,     0., -3000.], dtype=float32),
    jerk=array([0., 0., 0.], dtype=float32),
    thrust=array([0., 0., 0.], dtype=float32)
)
```
Isn't this legitimately setting position (0, 0, -10), velocity (0, 0, 0), and acceleration (0, 0, -3000) setpoints? Here, https://docs.px4.io/master/en/flight_modes/offboard.html#copter-vtol, "Position setpoint and velocity setpoint and acceleration" is one of the supported combinations.

Jaeyoung-Lim, 2022-01-09:
> Isn't this legitimately setting position (0, 0, -10), velocity (0, 0, 0), and acceleration (0, 0, -3000) setpoints?

It is a legitimate setpoint, but not a feasible one (I am assuming your vehicle is not capable of generating 3000m/s^2

> Still, I think the above should really be clarified in the documentation. Many places in the documentation, offboard mode is listed as one of the flight modes.

It is a flight mode, but it is in a sense a free-wheeling mode since the firmware stops making decisions and the user injecting setpoint is now responsible for generating all the setpoints and ensure the required safety for it. This is useful for development and potentially unsafe.

> Also, the offboard section describes the mode as "Offboard mode is primarily used for controlling vehicle movement and attitude, and supports only a very limited set of MAVLink messages (more may be supported in future).", not mentioning anywhere that those parameters will be ignored.

If there is something unclear in the documentation, please feel free to contribute directly in https://github.com/PX4/PX4-user_guide This will greatly help the community who will struggle with the same questions as you had. IMHO, It is hard to mention for each setpoint combination which parameters will be ignored. Currently the setpoints in principle can have 2^9 combinations but PX4 only supports a subset of those (which is clearly defined in the documentation)

Jaeyoung-Lim, 2022-08-15:
Closing as stale, @squizz617 please feel free to reopen

closed 2022-08-15 by Jaeyoung-Lim