PX4/PX4-Autopilot #18856: crash during landing / vehicle does not respond

NicolasM0, 2021-12-15:
**Describe the bug**
I was testing a custom version build upon recent master (95222baa332926983218c2915ebef3665552a8c1).
This version has some custom changes, principally a new module estimating a target position and a flight stack ran in place of "follow me" flight task. I suppose that this issue is not linked to these changes but don't know how to be sure.

My UAV was flying successfully and I entered RTL mode. It starts to land but when it hit the ground, it jumped a bit  (30cm ?) and fall to the ground again. On the ground, motors keep turning for about 3 min. I engaged the kill switch without effect. No RC command seems to do anything. My wifi telemetry link was lost (CUAV wifi link; connected to the wifi but not getting any data). I ended this by unplugging the battery. Luckily my vehicle has no damage as it happened just above the ground and motors kept a command low enough.

Looking into the log (https://review.px4.io/plot_app?log=714ae1ff-6830-4071-a93f-92d96127dae3), I can see that I have lost the GPS, the RC, the commander: these topics are not published anymore.

The log shows a "[logger] Writing watchdog data" message. I suppose it is linked to this:
```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                   92702  0.000   256/  512   0 (  0)  READY  3
   1 hpwork                          0  0.000   332/ 1264 249 (249)  w:sig  3
   2 lpwork                          0  0.000  1092/ 1616  50 ( 50)  READY  3
   3 init                            0  0.000  2260/ 2928 100 (100)  w:sem  3
   4 wq:manager                      0  0.000   596/ 1256 255 (255)  w:sem  4
 952 log_writer_file                 6  0.674   628/ 1176 255 (255)  w:sem  4
  55 wq:lp_default                   0  0.000  1172/ 1920 205 (205)  READY  4
  62 wq:hp_default                  67  6.722  1244/ 1904 237 (237)  READY  4
  79 dataman                         0  0.000   836/ 1208  90 ( 90)  w:sem  4
 362 wq:SPI1                       144 14.426  1676/ 2336 253 (253)  READY  4
 374 wq:SPI4                         3  0.334   996/ 2336 250 (250)  w:sem  4
 377 wq:I2C3                         3  0.317  1044/ 2336 244 (244)  w:sem  4
 557 wq:I2C2                        11  1.136  1044/ 2336 245 (245)  w:sem  4
 584 wq:nav_and_controllers         62  6.250  1572/ 2240 242 (242)  w:sem  4
 587 wq:rate_ctrl                  105 10.580  1620/ 3000 255 (255)  w:sem  4
 588 wq:INS0                        55  5.516  4364/ 6000 241 (241)  w:sem  4
 614 wq:INS1                        60  6.066  4364/ 6000 240 (240)  w:sem  4
 617 wq:INS2                        56  5.681  4364/ 6000 239 (239)  READY  4
 624 commander                       0  0.000  1324/ 3224 140 (140)  READY  5
 709 gps                             0  0.000  1204/ 1680 205 (205)  READY  4
 764 mavlink_if0                     0  0.000  1876/ 2728 100 (100)  READY  4
 765 mavlink_rcv_if0                 0  0.000  1612/ 4552 175 (175)  READY  4
 818 wq:UART6                      367 36.690  1120/ 1632 228 (228)  READY  4
 947 logger                         28  2.847  2796/ 3648 255 (255)  RUN    4
 881 navigator                       0  0.000  1164/ 1920 105 (105)  READY  6

Processes: 25 total, 13 running, 12 sleeping
CPU usage: 97.24% tasks, 2.76% sched, 0.00% idle
DMA Memory: 5120 total, 1536 used 2048 peak
Uptime: 299.238s total, 92.703s idle
```
wq:UART6 has very high CPU consumption  ` 818 wq:UART6                      367 36.690  `

Some other tasks don't seem to run anymore.

**To Reproduce**
I have not reproduced this in SIH and don't when to try again on a true vehicle before I understand what was happening.

**Drone (please complete the following information):**
The drone is a quadcopter built with a CUAV v5+ autopilot.
The ESC is a T-motor F55 BLHeli running in dshot mode (so SYS_USE_IO is disabled)
the RC is a Frsky Taranis and the receptor is plugged into dsm/sbus connector.

NicolasM0, 2021-12-15:
Thank you for this information. 
I had the issue less than 5min after startup...
Tell me if I can do something to help finding the issue

dagar, 2021-12-15:
> Thank you for this information. I had the issue less than 5min after startup... Tell me if I can do something to help finding the issue

Let me know if you're able to reproduce it. So far I'm not having any luck.

dagar, 2021-12-16:
@NicolasM0 if you're able to reproduce reliably please give https://github.com/PX4/PX4-Autopilot/pull/18861 a try.

Kim-zhi-jiang, 2021-12-16:
@NicolasM0  I also found this problem in the sim, looking forward repair [https://github.com/PX4/PX4-Autopilot/issues/18718](url)

dagar, 2021-12-23:
@NicolasM0 if you're able to reproduce please try https://github.com/PX4/PX4-Autopilot/pull/18893.

I'll keep this issue open until we have additional data.

NicolasM0, 2022-03-23:
Not reproduced...

closed 2022-03-23 by NicolasM0

dagar, 2022-03-23:
> Not reproduced...

Thanks for the followup @NicolasM0.