PX4/PX4-Autopilot #18873: Error " invalid -j parameter" while running "make px4_sitl_default gazebo"

mollaemin, 2021-12-18:
I am running command "make px4_sitl_default gazebo" on Ubuntu18.04(VM) to work with px4 and gazebo, I install all dependencies as shown in http://docs.px4.io/master/en/simulation/gazebo.html however, I am getting the below error every time. I didn't see anybody who takes this kinda failure all forums, so I thought somebody can help me here.

[837/841] Performing build step for 'sitl_gazebo' ninja: fatal: invalid -j parameter FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build cd /home/mollaemin/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/mollaemin/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 0 ninja: build stopped: subcommand failed. Makefile:225: recipe for target 'px4_sitl_default' failed make: *** [px4_sitl_default] Error 1

Kim-zhi-jiang, 2021-12-18:
Check your environment, 100% the version of each package, then will be OK

mollaemin, 2021-12-18:
> Check your environment, 100% the version of each package, then will be OK

As I mentioned above I follow all instructions given on the website.I don't know what can I do more. I am unfamiliar to ubuntu and all happenings behind the scenes

Kim-zhi-jiang, 2021-12-18:
This is open source Linux, changes every day, you need to solve environment by yourself

julianoes, 2021-12-20:
Duplicate of #18738.

closed 2021-12-20 by julianoes