PX4/PX4-Autopilot #18894: Build fail .elf file

ttropcic, 2021-12-23:
Hi,
Having problem with building the code.

Command:
$ make px4_fmu-v3_default

Generating C implementation in directory /cygdrive/c/PX4/home/PX4-Autopilot/build/px4_fmu-v3_default/mavlink/common
Generating C implementation in directory /cygdrive/c/PX4/home/PX4-Autopilot/build/px4_fmu-v3_default/mavlink/minimal
Copying fixed headers for protocol 2.0 to /cygdrive/c/PX4/home/PX4-Autopilot/build/px4_fmu-v3_default/mavlink
[1088/1090] Linking CXX executable px4_fmu-v3_default.elf
FAILED: px4_fmu-v3_default.elf
: && /cygdrive/c/PX4/toolchain/gcc-arm/bin/arm-none-eabi-g++.exe  -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -DNDEBUG  --specs=nosys.specs @CMakeFiles/px4.rsp  -o px4_fmu-v3_default.elf  && :
c:/px4/toolchain/gcc-arm/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file /cygdrive/c/PX4/home/PX4-Autopilot/boards/px4/fmu-v3/nuttx-config/scripts/script.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make: *** [Makefile:225: px4_fmu-v3_default] Error 1

I don't understand the issue. That script that it says isn't there, is there. So if somebody could maybe suggest a solution, I would be very grateful.
Thanks!

dagar, 2021-12-26:
I think it's a bit convoluted because the arm-none-eabi toolchain is a proper windows build and needs a windows path (c:). 

We probably need to figure out a proper Cygwin path here for the linker script. https://github.com/PX4/PX4-Autopilot/blob/72065c3d71c2eec5d682a95a0e0796086bf7b5f2/platforms/nuttx/CMakeLists.txt#L128

This is what we had previously. https://github.com/PX4/PX4-Autopilot/blob/2e8918da66af37922ededee1cc2d2efffec4cfb2/platforms/nuttx/CMakeLists.txt#L128

dagar, 2021-12-27:
@ttropcic could you try this? https://github.com/PX4/PX4-Autopilot/pull/18914