PX4/PX4-Autopilot #18904: Issue compiling 1.11.3 with Kconfig

zlowell2, 2021-12-23:
Hello - I am currently trying to compile an iteration of 1.11.3 PX4 on my macbook pro with an M1 chip and I get the following error: 

[image]

What is strange is I am able to compile 1.12.3 just fine - I am assuming it is an issue with nuttx and the kconfig-frontends install via brew? 

Any advice would be great - currently need the 1.11.3 iteration to compile - tried uninstalling brew then reinstalling and running ./Tools/setup/macos.sh, but am still provided with the same error whenever I compile.

dagar, 2021-12-26:
Sorry, I'm not sure what's going on here, but you can always use the specific container version that corresponds to a release. https://github.com/PX4/PX4-Autopilot/blob/72065c3d71c2eec5d682a95a0e0796086bf7b5f2/Tools/docker_run.sh#L7

Any reason you need to be on v1.11.3 instead of current PX4 (currently v1.12, but soon to be v1.13)?

zlowell2, 2021-12-26:
Gotcha - Ill take a look into the docker container, thanks! Reason for using this depreciated version is because some customizations I have implemented in 1.11.3 and have yet to move to 1.12.3. Because of this I tried getting base 1.11.3 to compile before customized firmware.

hhenryy, 2021-12-26:
I assumed you have cherry picked the commits to add the Kconfig feature in v1.11.3? Those commits are dependent on certain features since that PR is rebased on master. 

So if you cherry picked the commits you need to backport some of the changes those commits will make on your v1.11.3

I have successfully done this in v1.12.3.

zlowell2, 2021-12-27:
@hhenryy It appears I am getting this error too when I compile with the base 1.11.3 that has NOT been cherry picked. Just doing a git clone, then checking out 1.11.3, followed by submodule update and a compilation still leads me to the Kconfig error. Any idea why that would be happening?

hhenryy, 2021-12-27:
after you checkout to v1.11.3 with `git checkout v1.11.3` you need to run the following two commands:
```
git submodule sync --recursive
git submodule update --init --recursive
```

zlowell2, 2021-12-27:
Here is my output after trying that:

-- PX4 version: v1.11.3
-- PX4 config file: /Users/zach/PX4-Autopilot/boards/cubepilot/cubeorange/default.cmake
-- PX4 config: cubepilot_cubeorange_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.3.1
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Enabling double FP precision hardware instructions
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.9", minimum required is "3") 
-- build type is MinSizeRel
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-404-g395c8d3
-- Building and including cubepilot_io-v2_default
CMake Deprecation Warning at src/drivers/uavcan/libuavcan/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/drivers/uavcan/libuavcan/libuavcan/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /usr/local/bin/python3 (found version "3.9.9") 
-- Using C++11
-- Release build type: MinSizeRel
-- Building and including bootloader cubepilot_cubeorange_bootloader
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/zach/PX4-Autopilot/build/cubepilot_cubeorange_default
[  0%] git submodule platforms/nuttx/NuttX/nuttx
[  0%] Built target px4_config_file_target
[  0%] git submodule platforms/nuttx/NuttX/apps
[  0%] Generating uORB topic headers
[  0%] git submodule src/lib/ecl
[  0%] Built target git_nuttx
[  0%] Built target git_nuttx_apps
[  0%] Built target git_ecl
[  1%] Generating mixer_multirotor.generated.h
[  1%] Generating mixer_multirotor_normalized.generated.h
[  1%] Generating mixer_multirotor_6dof.generated.h
[  1%] Built target mixer_gen
[  1%] Built target mixer_gen_6dof
[  1%] Generating git version header
[  1%] git submodule src/drivers/uavcan/libuavcan
[  1%] git submodule src/drivers/gps/devices
[  1%] Built target git_gps_devices
[... 119 lines cut ...]
[  1%] Built target uorb_headers
[  3%] Built target nuttx_config_kconfig_target
[  4%] Built target nuttx_config_kconfig_target
[  6%] Built target nuttx_copy_config_dir_target
[  5%] Built target nuttx_copy_config_dir_target
[  5%] Copying NuttX compressed config bootloader and inflating (make olddefconfig)
[  6%] Copying NuttX compressed config nsh and inflating (make olddefconfig)
drivers/Kconfig:433: 'drivers/platform/Kconfig' not found (in 'source drivers/platform/Kconfig'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
drivers/Kconfig:433: 'drivers/platform/Kconfig' not found (in 'source drivers/platform/Kconfig'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
make[7]: *** [do_olddefconfig] Error 1
make[7]: *** [do_olddefconfig] Error 1
make[6]: *** [NuttX/nuttx/.config] Error 2
make[6]: *** Deleting file `NuttX/nuttx/.config'
make[5]: *** [NuttX/CMakeFiles/nuttx_config_target.dir/all] Error 2
make[6]: *** [NuttX/nuttx/.config] Error 2
make[6]: *** Deleting file `NuttX/nuttx/.config'
make[4]: *** [all] Error 2
make[5]: *** [NuttX/CMakeFiles/nuttx_config_target.dir/all] Error 2
make[3]: *** [external/Stamp/bootloader_firmware/bootloader_firmware-build] Error 2
make[4]: *** [all] Error 2
make[2]: *** [src/systemcmds/bl_update/CMakeFiles/bootloader_firmware.dir/all] Error 2
make[3]: *** [external/Stamp/px4io_firmware/px4io_firmware-build] Error 2
make[2]: *** [src/drivers/px4io/CMakeFiles/px4io_firmware.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [cubepilot_cubeorange_default] Error 2

dagar, 2021-12-27:
@zlowell2 it's looks like something is failing within the NuttX level kconfig.

Can you run `make distclean`, `make submodulesclean`, and then try again? This is to make sure there's absolutely nothing dirty within the submodules.

If you're still failing after that we need to go into the actual NuttX build and get proper error output.

``` Console
$ VERBOSE=1 make cubepilot_cubeorange_default
```

Then within the build directory look for `nuttx_context.log`, `nuttx_olddefconfig.log`, etc.

zlowell2, 2021-12-27:
This is what is within the nuttx_olddefconfig.log file: 

Create .version
LN: include/arch to arch/arm/include
LN: arch/arm/src/board to /Users/zach/PX4-Autopilot/build/cubepilot_cubeorange_default/NuttX/nuttx/../nuttx-config//src
LN: arch/arm/src/chip to arch/arm/src/stm32h7
LN: /Users/zach/PX4-Autopilot/build/cubepilot_cubeorange_default/NuttX/nuttx/drivers/platform to /Users/zach/PX4-Autopilot/build/cubepilot_cubeorange_default/NuttX/nuttx/../nuttx-config/drivers
LN: include/arch/board to /Users/zach/PX4-Autopilot/build/cubepilot_cubeorange_default/NuttX/nuttx/../nuttx-config//include
LN: include/arch/chip to arch/arm/include/stm32h7
LN: platform/board to /Users/zach/PX4-Autopilot/build/cubepilot_cubeorange_default/NuttX/apps/platform/dummy
DEBUG: kconfiglib kconfig-conf wrapper, arguments: --olddefconfig Kconfig

As for what verbose is outputting: 

[PX4-1.11.3-output.txt](https://github.com/PX4/PX4-Autopilot/files/7781647/PX4-1.11.3-output.txt)

zlowell2, 2021-12-29:
I get the same error when running inside a docker container that has been platformed to emulate x84/i386

'make cubepilot_cubeorange_default' results in 

boards/Kconfig:2303: 'boards/arm/lpc17xx_40xx/lincoln60/Kconfig' not found (in 'source "boards/arm/lpc17xx_40xx/lincoln60/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.

tpwrules, 2022-02-17:
This seems to possibly be the fault of a newer CMake. I get this same problem when using CMake 3.18.2 (as part of NixOS 20.09), but when I downgraded to CMake 3.16.3 (as part of NixOS 20.03) then everything built fine. What version of CMake are you using? Can you try downgrading it?

I did not bother investigating why this issue occurs.