PX4/PX4-Autopilot #18918: Run SITL without estimator?

heleidsn, 2021-12-28:
Hi developers,

I am doing some RL stuff using ROS+PX4 SITL. So I need to reset the world after each episode. This will cause an EKF error. I am wondering if it is possible to run SITL using the ground truth pose directly from the gazebo without any estimators? 

Thanks!

hhenryy, 2021-12-28:
I used PX4 alongside this gazebo plugin https://github.com/jsbruglie/gap for domain randomization.

I wrote a bash script which restarts the simulation from scratch after the simulation finished or something failed(easily done with ROS). This results in the EKF being re-initialized and there are no prior learnings from the previous simulation run. This ensure that simulations are isolated from each other

heleidsn, 2021-12-28:
Hi @hhenryy , thanks for your quick reply. Actually, I don't want to restart the simulation at all as it will take too much time. I am using `rosservice call /gazebo/reset_world` to restart the simulation. In lockstep mode, it takes about 15s for ekf2 to converge to the new location. One of the solutions is that `ekf2 stop` then `ekf2 start `, but there is some problems sometimes.

Jaeyoung-Lim, 2021-12-28:
@heleidsn What is the advantage of simulating the px4 firmware when trying to learn from simulation?

If your input states are just input states, there is no reason to run the px4 firmware for learning. You can simply simulate this through the dynamics simulation without Software-In-The-Loop

hhenryy, 2021-12-28:
I used it as transfer learning to estimate disturbances acting on the vehicle.

heleidsn, 2021-12-28:
Hi @Jaeyoung-Lim ,

Thanks for your reply. I have already seen your answer at [here](https://discuss.px4.io/t/how-to-reset-px4-sitl-state/18399/6). 

Yeah, I totally agree with you about using dynamics simulation without SITL for learning stuff. However, I didn't find a simple dynamics simulator with depth camera simulation and a good controller interface. Should I code it by myself or could you please tell me some good dynamics simulators? 

I have used PX4 for years and it will be also used for the real test. That's why I choose PX4 SITL as the simulator. 

Finally, is it possible for PX4 sitl to use the ground truth data for controller rather than estimator? I don't know if there are any technical difficulties and this configuration can also be used to verify the controller. But I didn't find any solutions for this.

Jaeyoung-Lim, 2021-12-28:
> Yeah, I totally agree with you about using dynamics simulation without SITL for learning stuff. However, I didn't find a simple dynamics simulator with depth camera simulation and a good controller interface. Should I code it by myself or could you please tell me some good dynamics simulators?

I suppose you mean by PX4 SITL as PX4 SITL Gazebo (PX4 supports many different simulators). My suggestion is to not use PX4 SITL + Gazebo but just Gazebo. You are not gaining anything by simulating PX4 side of the simulator and this limits your ability to randomize your dynamics, which is critical for doing sim2real.

> I have used PX4 for years and it will be also used for the real test. That's why I choose PX4 SITL as the simulator.

PX4 SITL is NOT a dynamics simulator. PX4 SITL is just simulating PX4, which can talk to different simulators such as gazebo/airsim/jmavsim etc. You can read up on how this works in http://docs.px4.io/master/en/simulation/

> Finally, is it possible for PX4 sitl to use the ground truth data for controller rather than estimator? I don't know if there are any technical difficulties and this configuration can also be used to verify the controller. But I didn't find any solutions for this.

I am not sure what you are trying to achieve by this. You can forward ground truth data to the estimator to use this as if it is coming from a external position source with the External Vision pipeline. But if you turn off the estimator, since PX4 has a lot of safety features built in depending on the estimator state you would need to build an additional state machine that forwards groundtruth data to the controllers.

heleidsn, 2021-12-28:
@Jaeyoung-Lim Many thanks. I realized the difficulties of turning off the estimator. If I use only Gazebo to simulate the dynamics, what controller can I use? I know AirSim has a simple controller for quadrotors. Are there any similar things? I have to simulate the 6DOF because the camera should be fixed on the quadrotor, and the policy output will be some high-level command such as velocity setpoint. Btw, is there any controller interface for PX4 similar to the mavros but without ROS, for example, control directly using python?

Jaeyoung-Lim, 2021-12-28:
@heleidsn What kind of controller do you need and why do you need a controller in the first place if you are trying to learn a policy in RL?

If you are learning high level (e.g. velocity commands) from depth images this means that it doesnt even depend on any of the controller behavior. If you are trying to run a low level controller through RL then you dont need any of the cascaded control structure

heleidsn, 2021-12-29:
Hi @Jaeyoung-Lim , I am going to learn a high-level policy to output velocity commands from depth images and some states (e.g. current position and velocity). The thing is that I don't really want to re-write the quadrotors' dynamics. I did a little bit of search, there are some dynamics modules that seems useful, such as [Quadcopter_SimCon](https://github.com/bobzwik/Quadcopter_SimCon). 

However, I still want to make sure if px4 has any support for simulating the controller only using ground truth data as the estimator which is similar to the [AirSim simple flight](https://microsoft.github.io/AirSim/simple_flight/)? Many thanks for your reply!

heleidsn, 2022-01-19:
Finally, I write a simple dynamics model to generate the position and attitude and train the policy with AirSim. It runs well. This issue will be closed. Thanks.

closed 2022-01-19 by heleidsn