PX4/PX4-Autopilot #18924: Cannot Build PX4 sitl on  fresh cloned PX4-Autopilot

pbecchi, 2021-12-30:
I got this error:
```
~/PX4-Autopilot$ sudo HEADLESS=1 make px4_sitl jsbsim_quadrotor_x
-- PX4 version: v1.13.0-alpha1-3911-g5fc7e551ab
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- PX4 config file: /home/pbecchi/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ROMFS: ROMFS/px4fmu_common
DEBUG1 logical cores detected and 645 megabytes of memory available.
		Limiting sitl_gazebo and simulation-ignition concurrent jobs to 0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pbecchi/PX4-Autopilot/build/px4_sitl_default
[1/853] Generating mixer_multirotor_normalized.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor_normalized.generated.h 
cd /home/pbecchi/PX4-Autopilot/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --normalize -f /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/pbecchi/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor_normalized.generated.h
Illegal instruction (core dumped)
[2/853] Generating uORB topic headers
ninja: build stopped: subcommand failed.
Makefile:230: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
```
I don't know the reason but 2 weeks ago I could `make PX4_sitl` with no errors. I have deleted the directory and re-cloned the repo , but this error don't go away. 
What else is changed? May be some dependencies.......

Jaeyoung-Lim, 2021-12-30:
@pbecchi I don't think you should use sudo to build the firmware.

pbecchi, 2021-12-30:
Exactly same error  without sudo.
I wonder why 2 weeks ago was it was working.......

pbecchi, 2021-12-30:
I did some other tests.
Tryed to run mixer_multirotor.py with python3 interpreter.
I got same error:
```
cd src/lib/mixer/MultirotorMixer
python3 mixer_multirotor.py
Illegal instruction (core dumped)
```
I know nothing about python .... but i guess there is something wrong with it. 
How I can check it is OK?
I should reinstall it??
Any advice ??

pbecchi, 2021-12-31:
I have tried to reinstall python3 but I keep having same problem.
Finally I have fount a way around : 
I have modified src/lib/mixer/MultirotorMixer/CMakeLists.txt
defining locally PYTHON_EXECUTABLE /usr/bin/python2
Now building go ahead up to another similar error:
`-- Configuring done
-- Generating done
-- Build files have been written to: /home/pbecchi/PX4-Autopilot/build/px4_sitl_default
[4/852] git submodule src/drivers/gps/devices
[6/852] git submodule src/modules/mavlink/mavlink
[239/852] git submodule Tools/jsbsim_bridge
[240/852] Generating Mavlink developme...ssage_definitions/v1.0/development.xml
Illegal instruction (core dumped)
FAILED: mavlink/development/development.h 
cd /home/pbecchi/PX4-Autopilot/build/px4_sitl_default/src/modules/mavlink && /usr/bin/python3 /home/pbecchi/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py --lang C --wire-protocol 2.0 --output /home/pbecchi/PX4-Autopilot/build/px4_sitl_default/mavlink /home/pbecchi/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
`
Why this appends?? What is wrong with python3?
I don't have any idea of what could be!!Anybody can help me with this??

pbecchi, 2022-01-02:
Python dependencies were screwed up....I had to reinstall Ubuntu 18.04.I have installed all python3 dependencies required.
Now `make px4_sitl-default` compleate with no errors  .
But `make px4_sitl jsbsim_quadrotor_x` give me these errors:

`-- Found MAVLink: /home/pbecchi/PX4-Autopilot/build/px4_sitl_default/mavlink (found version "2.0") 
[  4%] Built target perf
[  6%] Built target work_queue
[  6%] Built target geo
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:8 (find_package)

CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:8 (find_package)

CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:567 (message):
  Imported targets and dependency information not available for Boost version
  (all versions older than 1.33)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:8 (find_package)

[  6%] Built target MixerBase
[  6%] Built target mavlink_c_generate
CMake Error at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find the requested Boost libraries.

  Unable to find the Boost header files.  Please set BOOST_ROOT to the root
  directory containing Boost or BOOST_INCLUDEDIR to the directory containing
  Boost's headers.
Call Stack (most recent call first):
  CMakeLists.txt:8 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find TinyXML (missing: TinyXML_LIBRARY TinyXML_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindTinyXML.cmake:65 (find_package_handle_standard_args)
  CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pbecchi/PX4-Autopilot/build/px4_sitl_default/build_jsbsim_bridge/CMakeFiles/CMakeOutput.log".
See also "/home/pbecchi/PX4-Autopilot/build/px4_sitl_default/build_jsbsim_bridge/CMakeFiles/CMakeError.log".
platforms/posix/CMakeFiles/jsbsim_bridge.dir/build.make:105: recipe for target 'external/Stamp/jsbsim_bridge/jsbsim_bridge-configure' failed
make[4]: *** [external/Stamp/jsbsim_bridge/jsbsim_bridge-configure] Error 1
CMakeFiles/Makefile2:45688: recipe for target 'platforms/posix/CMakeFiles/jsbsim_bridge.dir/all' failed
make[3]: *** [platforms/posix/CMakeFiles/jsbsim_bridge.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[  6%] Built target component_general_json
[  7%] Built target HelicopterMixer
[ 27%] Built target uorb_msgs
CMakeFiles/Makefile2:15293: recipe for target 'platforms/posix/CMakeFiles/jsbsim_quadrotor_x.dir/rule' failed
make[2]: *** [platforms/posix/CMakeFiles/jsbsim_quadrotor_x.dir/rule] Error 2
Makefile:3215: recipe for target 'jsbsim_quadrotor_x' failed
make[1]: *** [jsbsim_quadrotor_x] Error 2
Makefile:230: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 2
`
Can you help me with this??

Jaeyoung-Lim, 2022-01-02:
> Imported targets and dependency information not available for Boost version

You are missing boost as a dependency

pbecchi, 2022-01-02:
Yes thanks fixed...another error:
`-- /home/pbecchi/PX4-Autopilot/build/px4_sitl_default/mavlink
CMake Error at CMakeLists.txt:49 (pkg_check_modules):
  Unknown CMake command "pkg_check_modules".

[  6%] Built target mavlink_c_generate

-- Configuring incomplete, errors occurred!
`
Something else is missing.....

Jaeyoung-Lim, 2022-01-03:
@pbecchi Have you tried following the toolchain installation instructions? If you have followed the instructions these cases should not have happened

pbecchi, 2022-01-03:
Finally I have run following commands:
`
 make px4_sitl_default install_python_requirements

sudo apt-get install libeigen3-dev
sudo apt-get install libtinyxml-dev
sudo apt-get install pkg-config
sudo apt-get install libgstreamer1.0-dev
sudo apt-get install libfcl-dev
sudo apt-get  install cmake libboost-all-dev`

And now finally it builds `make px4_sitl jsbsim_dj_f450`
End of this story will be on  github.com/JSBSim-Team/jsbsim issue#540

closed 2022-01-03 by pbecchi